littlefs-project/littlefs #156: Making littlefs thread-safe?

rbphilip, 2019-03-14:
Hey! I just found littlefs and am porting it this morning to a FreeRTOS system with 128MB NOR flash. 

I see that littlefs doesn't have any sort of thread-safe mechanism built in. That's fine and I've seen references to wrapping littlefs calls with semaphores. Also fine. 

The question is *which* calls need to be wrapped. open/close/read/write/??

Anything that has a prototype in lfs.h? That'll likely be my first stab at it, but perhaps this is an issue that someone else has resolved in an elegant manner..

Thanks for any insights.

FreddieChopin, 2019-03-14:
> The question is which calls need to be wrapped. open/close/read/write/??

Everything which is an user-facing public API.

For an example see this port:
https://github.com/DISTORTEC/distortos/tree/master/source/FileSystem/littlefs
https://github.com/DISTORTEC/distortos/tree/master/include/distortos/FileSystem/littlefs

rbphilip, 2019-03-14:
Unfortunately I have zero experience with C++. Is "const std::lock_guard<LittlefsFile> lockGuard {*this};" a semaphore "take"? 

Is this "return" the semaphore "give" "return {{}, ret};"

FreddieChopin, 2019-03-14:
> Is "const std::lock_guard lockGuard {*this};" a semaphore "take"?

Yes.

> Is this "return" the semaphore "give" "return {{}, ret};"

Not exactly. Any possible exit from this function will destroy this `lockGuard` object, which will cause its destructor to be executed, which in turn releases the mutex. Generally any exit from this function causes automatic release of the mutex.

Please also don't confuse mutex and semaphore - the fact that in FreeRTOS they are "almost the same thing" (one of many stupid things in FreeRTOS...) doesn't mean that they are meant for the same purpose.

rbphilip, 2019-03-14:
Thanks. 

It appears that there is a different mutex used for directories than for files. Do you perhaps know if this is true (again, I don't really grok C++) and if so, why?

FreddieChopin, 2019-03-14:
There are no separate mutexes for file and directory objects - there is only one recursive mutex per file system and locking the file or directory is achieved by locking the mutex in the associated file system.

`lockGuard` is only a RAII (look it up) object which locks/unlocks the mutex/object that you provide. The "thing" you provide just has to have `lock()` and `unlock()` functions which will be called in constructor and destructor respectively.

Here's the only mutex in these clases:
https://github.com/DISTORTEC/distortos/blob/master/include/distortos/FileSystem/littlefs/LittlefsFileSystem.hpp#L290

Note that there is no such mutex in file or directory classes. If you lock the file manually this calls `lock()` of associated file system:
https://github.com/DISTORTEC/distortos/blob/master/source/FileSystem/littlefs/LittlefsFile.cpp#L100
which in turn just locks the common mutex:
https://github.com/DISTORTEC/distortos/blob/master/source/FileSystem/littlefs/LittlefsFileSystem.cpp#L307

But you don't have to manually lock anything. Each function does its own locking (with the `std::lock_guard`). The `lock()` and `unlock()` functions are required only if you want to perform several operations "atomically", for example you need to do 10 writes back-to-back and don't want any write from other thread to mix things up (for example you are writing to a shared log file but you cannot write the complete entry with one call and need several of them).

rbphilip, 2019-03-14:
Thanks. C++ continues to seem unnecessarily obscure and unnecessarily complex to me. I suppose that comes from learning to program back when "C" was a new language. :)  

Can I distill down that what I need to do for each of the lfs externally visible functions to something like this:

int my_getsize(
{
    xSemaphoreTake( fs_mutex,0xffffffff);
  int size = lfs_file_size(&fileSystem_.fileSystem_, &file_);
   xSemaphoreGive( fs_mutex );
   return size;
}

FreddieChopin, 2019-03-14:
Yes.

rbphilip, 2019-03-14:
Thanks for the help.

geky, 2019-04-01:
Sounds like everything got sorted out here? Thanks @FreddieChopin! Let me know if this issue needs to be reopened.

closed 2019-04-01 by geky

rbphilip, 2019-04-01:
Yes - everything is working fine. As a future feature request it would be a lot easier to do thread-safe if there was the possibility of a "get a mutex" and "release a mutex" function configuration, and LittleFS called them itself at the appropriate places.

geky, 2019-04-01:
Not a bad idea, assuming 1. out of the box had default stubs (similar to lfs_malloc) and 2. stubs did not increase existing code size.

Unfortunately, this is low priority at the moment. I'd be happy to take in a PR, but I think there's some bigger issues to take care of first.

henrygab, 2019-12-19:
This is just an update of @FreddieChopin 's links:

#### Mutex:
https://github.com/DISTORTEC/distortos/blob/79062407feef8b80b28929610f4eb4e13e467b29/include/distortos/FileSystem/Littlefs1FileSystem.hpp#L309

#### File/Directory's `lock()`:
https://github.com/DISTORTEC/distortos/blob/746afae4c4d8485c9aafc572364861c6ac6de223/source/FileSystem/littlefs1/Littlefs1File.cpp#L94-L97

#### Filesystem-wide `lock()`:
https://github.com/DISTORTEC/distortos/blob/746afae4c4d8485c9aafc572364861c6ac6de223/source/FileSystem/littlefs1/Littlefs1FileSystem.cpp#L265-L269

johnnytolengo, 2021-09-17:
Hello  @rbphilip, could you provide us an example of the wrapper you did?  I have the same situation here and I have not clue how to do that thread-safe wrapper.

Thank you

rbphilip, 2021-09-17:
Sorry. That was several clients ago.

But basically create wrapper functions for all the littlefs calls, and 
use a semaphore to "lock" access to the littlefs library.. something 
like below.

somewritefunction (data, size)
{
      take semaphore
      call littlefs function to write
      release semaphore
}

johnnytolengo, 2021-09-17:
yes, but that is already build in look here: https://github.com/lorol/LITTLEFS/blob/master/src/esp_littlefs.c#L662

I think the only problem is around the file descriptor when the file is opened multiple times: https://github.com/joltwallet/esp_littlefs/pull/19

I am not sure if that is already solved in the https://github.com/lorol/LITTLEFS

I it's all about thread-safty  :(

rbphilip, 2021-09-17:
> is opened multiple times: joltwallet/esp_littlefs#19

It's been years since I even thought of littlefs.

FreddieChopin, 2021-09-17:
@rbphilip :

> It's been years since I even thought of littlefs.

Have you maybe found some nice alternatives to it? (;

rbphilip, 2021-09-17:
Sorry, no.

It worked well for the client at the time but I haven't had need for a 
file system on any projects since then.

amgross, 2025-01-21:
Just to mention to whoever get to here, there is now thread safe embedded in littlefs when defining LFS_THREADSAFE