littlefs-project/littlefs #157: cppcheck warning due to if (true) inside while (true)

armandas, 2019-03-15:
The following code in lfs.c causes a cppcheck warning

> Identical inner 'if' condition is always true (outer condition is 'true' and inner condition is 'true').

```
    // begin loop to commit compaction to blocks until a compact sticks
    while (true) {
        if (true) {
```

`if (true)` appears in several places in littlefs. Is this done for debugging purposes or some other reasons? Is it possible to change the implementation to avoid the warning?

geky, 2019-04-01:
The `if (true) {` is to avoid a warning about goto bypassing variable initialization. If you replace it with just `{` does the warning go away?

armandas, 2019-04-02:
I did not get any compiler warnings with or without `if (true)` (only `error: redefinition of 'err'`). The error does not happen when `if (true)` is replaced by `{`.

I am using GCC for ARM, V6.3.1.

geky, 2019-04-09:
Thanks for raising an issue, I pushed up a fix on the v2-alpha branch (https://github.com/ARMmbed/littlefs/pull/85).

Related https://github.com/ARMmbed/littlefs/issues/155

armandas, 2019-04-10:
Thank you. I can confirm that the cppcheck warnings are gone.

closed 2023-01-23 by armandas