littlefs-project/littlefs #159: Always formatting, bad blocks

kazola, 2019-03-22:
Dear,

just started working with littleFS on my ARM device. I use a 64 Kbytes (8K * 8) FRAM SPI memory. I was able to build your library. And I was able to run the code. 
SPI communication works fine, already tested.
However, I must be doing something wrong, my output is:

[Cortex_M4_0] lfs error:499: Corrupted dir pair at 0 1
lfs error:2238: Invalid superblock at 0 1
lfs debug:623: Bad block at 3
...
lfs debug:623: Bad block at 31
lfs warn:317: No more free space 32
lfs error:499: Corrupted dir pair at 0 1
lfs error:2238: Invalid superblock at 0 1

Curiously, there is no bad block number 2 :)

Ok, this is what I am trying to do with littlefs, very simple at this point!
`    // APP: manage file system, allow writing
    bool transfer_ok;
    transfer_ok = fram_write_enable();
    int err = lfs_mount(&lfs, &cfg);
    if (err) {
        lfs_format(&lfs, &cfg);
        lfs_mount(&lfs, &cfg);
    }
    transfer_ok = fram_write_disable();`

This is my configuration:
`const struct lfs_config cfg = {
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync  = user_provided_block_device_sync,

    .read_size = 256,
    .prog_size = 256,
    .block_size = 256,
    .block_count = 32,
    .lookahead = 32,
};`

These are my hooks:
`int user_provided_block_device_read(const struct lfs_config *c,
                                    lfs_block_t block,
                                    lfs_off_t off,
                                    void *buffer,
                                    lfs_size_t size)
{
    fram_read_memory((block * c->block_size + off), (uint8_t*)buffer, size);
    return 0;
}
int user_provided_block_device_prog(const struct lfs_config *c,
                                    lfs_block_t block,
                                    lfs_off_t off,
                                    const void *buffer,
                                    lfs_size_t size)
{
    fram_write_memory((block * c->block_size + off), (uint8_t*)buffer, size);
    return 0;
}
int user_provided_block_device_erase(const struct lfs_config *c,
                                     lfs_block_t block)
{
    fram_erase_memory(block * c->block_size);
    return 0;
}
int user_provided_block_device_sync(const struct lfs_config *c)
{
    return 0;
}`

And this is the code for the function called by hooks:
`bool fram_write_memory(uint32_t address,
                       uint8_t * buf,
                       uint32_t len_buf)
{
    spi_clean_buffers();
    if (len_buf > SPI_MSG_LENGTH -3) return false;
    masterTxBuffer[0] = FRAM_OPCODE_WRITE;
    masterTxBuffer[1] = (address >> 8) & 0xFF;
    masterTxBuffer[2] = (address >> 0) & 0xFF;
    memcpy(&masterTxBuffer[3], buf, len_buf);
    return spi_wrapper(len_buf + 3);
}
bool fram_read_memory(uint32_t address,
                      uint8_t * buf,
                      uint32_t len_buf)
{
    spi_clean_buffers();
    if (len_buf > SPI_MSG_LENGTH -3) return false;
    masterTxBuffer[0] = FRAM_OPCODE_READ;
    masterTxBuffer[1] = (address >> 8) & 0xFF;
    masterTxBuffer[2] = (address >> 0) & 0xFF;
    bool rv = spi_wrapper(len_buf + 3);
    memcpy(buf, masterRxBuffer, len_buf);
    return rv;
}
bool fram_erase_memory(uint32_t address)
{
    spi_clean_buffers();
    int len_buf = SPI_MSG_LENGTH;
    masterTxBuffer[0] = FRAM_OPCODE_WRITE;
    masterTxBuffer[1] = (address >> 8) & 0xFF;
    masterTxBuffer[2] = (address >> 0) & 0xFF;
    memset(&masterTxBuffer[3], 0x0, len_buf);
    return spi_wrapper(len_buf + 3);
}`

SPI code works fine so I must be doing something really wrong.
Please let me know your opinion! Also tried erasing with 0xFF, no luck.
Meanwhile, I will keep reviewing documentation, could not read all of it, yet.

FreddieChopin, 2019-03-22:
> memcpy(buf, masterRxBuffer, len_buf);

In here you copy the 3 bytes of header and 253 bytes of what you read from the memory, instead of the 256 bytes you read from the memory.

kazola, 2019-03-22:
Thanks @FreddieChopin !

Now it seems it can mount, and I added more of the example code, but the boot_count variable is always = 1 and it gives the single following error:

[Cortex_M4_0] lfs error:499: Corrupted dir pair at 3 2
lfs error:499: Corrupted dir pair at 3 2

Any idea?

kazola, 2019-03-22:
After formatting again and some adjust to the block layer, it works. Already counting boot_count file. Have the best weekend you all.

closed 2019-03-22 by kazola

techNiv1, 2020-09-11:
Dear @kazola,

can u please elaborate how you find solution for your problem?
or could you please provide me some pointers for the same?
Actually I am also getting the same issue.

> custom/lfs.c:1170:error: Corrupted dir pair at {0x0, 0x1}
custom/lfs.c:1970:debug: Bad block at 0x0
custom/lfs.c:1977:warn: Superblock 0x0 has become unwritable
custom/lfs.c:4248:trace: lfs_format -> -28