littlefs-project/littlefs #161: bad blocks cause LFS to crash

feistyz, 2019-03-24:
Hi,

We are testing how lfs deals with a lot of bad blocks. While somewhat of stress test situation we wanted to see what happens when we get multiple bad blocks in a row.

For the test, we are making our underlying flash driver return a fail on every other block erase. This, of course, causes lfs to consider the block as bad. However, it also causes lfs to crash at this assert:

LFS_ASSERT(pcache->block == 0xffffffff);

Here is the log output from lfs
[00:00:06.494,000] <info> lfs: lfs debug:617: Bad block at 4
[00:00:06.600,000] <info> lfs: lfs debug:640: Relocating 4 5 to 14 5
[00:00:06.744,000] <info> lfs: lfs debug:617: Bad block at 3
[00:00:06.850,000] <info> lfs: lfs debug:640: Relocating 3 2 to 15 2
[00:00:06.973,000] <info> lfs: lfs debug:617: Bad block at 1
[00:00:06.973,000] <warning> lfs: lfs warn:626: Superblock 1 has become unwritable

I suspect it is crashing because a Superblock becomes unwritable but that is just a speculation.

Our assumption is that since there are writable block lfs should keep working even though it will be slow. 

We are on the most recent version from master.

Thanks

techNiv1, 2020-09-11:
Hello @feistyz 
I am getting `custom/lfs.c:1970:debug: Bad block at 0x0
custom/lfs.c:1977:warn: Superblock 0x0 has become unwritable` this error, what am I doing wrong in my code?
where this issue lies?
please give me some pointers how to implement littleFS to my Flash.

feistyz, 2020-09-13:
There is not info in your post to say. There is a whole pile of things that can cause that.