littlefs-project/littlefs #162: reduce build size for read-only support

mjr-eyelock, 2019-03-26:
Our team is developing a custom firmware update feature with a custom bootloader. We are not using mbed OS; however, we are investigating the use of littlefs, as a filesystem abstraction to manage storage on a SPI-connected flash chip. Our main application will use the full read and write features of littlefs, but we are trying to reduce the size of our separate bootloader app by building in only those portions of littlefs needed to _read_ from the filesystem. So far, after a bit of basic hacking, we've only been able to reduce our bootloader application size by a few hundred bytes.

If perhaps some build flags were put into place that would produce build of a subset of the library that would support only read operations, is it feasible to expect the size difference to be more than just a few hundred bytes? If so, might this be added as a feature?

geky, 2019-04-12:
Hi @matthewjrichey, from previous experiments I would guess building littlefs readonly would get you to about 1/3rd the code size. Most of the code cost comes from managing writes.

If you remove the writing if statements from [lfs_file_flush](https://github.com/ARMmbed/littlefs/blob/0907ba7813173901bc9a7689e8781434629a3696/lfs.c#L2508-L2573) and [lfs_file_sync](https://github.com/ARMmbed/littlefs/blob/0907ba7813173901bc9a7689e8781434629a3696/lfs.c#L2586-L2623), the compiler should be able to garbage collect the write functions and give you a more accurate prediction.

Unfortunately this is relatively low priority for me at the moment, but I'd be happy to take in a PR.

thrasher8390, 2019-04-30:
I've been wondering the same thing. I'll have a look to see if I can get the build size down. If anyone else has tried this, it would be great to hear from you.

geky, 2019-07-26:
Just an FYI, @KRekha3 was able to modify littlefs for readonly and the resulting size was ~5KiB: https://github.com/ARMmbed/littlefs/issues/183