littlefs-project/littlefs #165: Semantics of LFS_ASSERT

zrnsm, 2019-04-10:
I managed to trip an assert in `lfs_file_sync` during a long running stress test:
```c
LFS_ASSERT(entry.d.type == LFS_TYPE_REG)
```
(which crashed my application because I had left asserts fully enabled). LFS came right back after a reboot.

I then noticed that it's possible to just disable asserts entirely.

Is `LFS_ASSERT` intended to mean "whatever happened should never ever happen and everything is hosed"? Or is there some conceivable reason that an assert could be non-fatal and you could continue in a sane way after encountering one?

geky, 2019-04-11:
Hi @zrnsm, asserts note program assumptions. If an assert fails, that means a programmer, somewhere, has done something wrong.

At this point the program was never designed to work without this assumption. Any masquerade that the program would be able to continue is pure luck. This safest thing to do in this situation is halt and report the bug.

If you're able to reproduce this assertion please do raise it as an issue, bug reports are very valuable for trying to build a stable piece of software.

> I then noticed that it's possible to just disable asserts entirely.

If the assumptions are not wrong, assertions don't change the behavior of the program, but they do have a size and runtime cost. Disabling assertions can be useful for decreasing the size of a program if it natively does not fit in your device. However it is riskier, as bugs may go unnoticed.

The best thing to do is run code with assertions, and only disable assertions if you really need the code size.

Hopefully that answers the question.

zrnsm, 2019-04-11:
Thanks for the clear answer, Chris. Will post again with more detail if I'm able to reproduce.

closed 2019-04-11 by zrnsm