littlefs-project/littlefs #166: Internal MCU flash configuration and performance tradeoffs

zrnsm, 2019-04-12:
[image]

@geky Noticed the huge PR that just got merged in and this part of your simulation results really got my attention. I'm trying to get a better understanding of the tradeoffs in a very resource constrained internal flash use case exactly like this. With small internal byte addressable pages like this is it possible for `block_count` to be very small, like say 1 or 2? How would that impact performance / wear leveling etc.? Maybe you could just comment generally on your sense of what the best practices are for this type of use case.

Up until now I've been using a separate flash translation layer to wear level across two internal flash pages because I wasn't getting the sense from the documentation that it made sense to use a configuration like this particular simulation, i.e. byte sized reads and writes, block size corresponding to an internal eraseable page size and possibly small `block_count`. Was that just an inaccurate perception on my part?

BTW, I really appreciate the swift response to my previous message and the great work on LFS overall.

zrnsm, 2019-04-14:
#11 Has some relevant discussion that has me leaning toward sticking with my LFS+FTL approach in order to have erase units that are much smaller than the physical erase unit. "On disk, each file uses 1 erase unit at minimum." That's the key limitation for my application.

If anyone has any other ideas/approaches for the internal flash use case, I'd be interested to hear them.

geky, 2019-04-14:
Hi @zrnsm, thanks :) 

littlefs on internal flash is worth a try. I'm not sure about is how LFS vs LFS+FTL compares _performance_ wise, but it may be worth it if code size is a concern.

It's a bit confusing right now because this changed recently. In v1, each file takes up 1 erase block, but in v2 this is no longer true. v2 introduced inline files, which allows multiple files to share a block pair.

v1 - 1 file per block  
v2 - 255 files per pair (this can be increased in the future)  

Better support for internal flash was a big driver for v2, which is especially useful for applications which use both internal and external flash but don't want to pay 2x the cost for two different filesystems.

It's important to note there are some annoying limitations with inline files.
- Must be &lt; cache_size, which means inline files are limited by RAM
- Limited to &le; 1023 bytes
- Limited to &le; 255 files per pair

(Note: I've reserved 2 bits in the tag format, we can use these to improve one of the two limits to allow either &le; 4095 bytes or &le; 1023 files per pair. I'm waiting to see which limit users run into more often before deciding which to increase.)

> With small internal byte addressable pages like this is it possible for block_count to be very small, like say 1 or 2? How would that impact performance / wear leveling etc.? 

So! With v2, at minimum littlefs needs 2 blocks. This is a fundamental minimum for any power-resilient filesystem, because at some point the block needs to be erased.

Wear leveling will work great. At this point, littlefs will behave as a sort of naive 2 block log, which means uniform static wear-leveling.

Performance may be more of a concern. littlefs relies on multi-block data structures for scalability, so it may not be as fast as other solutions out there. Nothing becomes exponential, but it does use a _O(n&sup2;)_ garbage collector.

Figuring out if this performance cost is actually a dealbreaker in the real world will probably require trying it out, I don't have much data on it. Usually the amount of data stored in internal flash is relatively small, so it may not be a problem.

> If anyone has any other ideas/approaches for the internal flash use case, I'd be interested to hear them.

You may also want to consider [SPIFFS](https://github.com/pellepl/spiffs). From what I understand it was designed with internal flash as the primary target.

zrnsm, 2019-04-14:
@geky Again thanks for the thorough discussion! Very useful. Going to mull this over and do some experiments with my application. I haven't had a chance to experiment much with v2 yet. 255 files with two blocks is probably plenty for my purposes. I did try SPIFFS in the past and had trouble getting it up and running.

geky, 2019-04-14:
Do let us know how it goes! Feedback is helpful.