littlefs-project/littlefs #168: crash within lfs_format

schoegmi, 2019-04-17:
I get a program crash during lfs_format()

This is my test:
1. format
2. mount
3. open, write, close one small file
4. umount
5. format again: this fails, because when lfs_format() calls lfs_dir_fetchmatch(), this will call cb() in line 928, which is NULL

I am using version v2.0.2 on a NOR flash with 4k blocks.
When I erase the first 2 blocks of the flash before calling lfs_format(), this error does not occur.

schoegmi, 2019-04-17:
Addendum: in contrast to the note in lfs_dir_fetch() the tag has the value 0.

schoegmi, 2019-04-17:
Sorry - was my fault.
I misunderstood the meaning of the block-sizes of the low level (*read) and (*write) vallback functions.

closed 2019-04-17 by schoegmi

geky, 2019-05-16:
Glad you figured it out 👍

Gspan93, 2019-09-17:
I am facing similar error as @ksstms, but in my case, it's a non-empty chip with first two blocks erased & have never seen lfs.
Even if I try to format the chip first, without mounting it (following the example in Readme) I have a NULL cb in lf_fs_traverse().
Here's the trace log for reference:
``
[CORTEX_M4_0] lfs_trace:3503: lfs_format(200033e8, ccf8 {.context=0, .read=9541, .prog=945d, .erase=94ef, .sync=9459, .read_size=16, .prog_size=256, .block_size=4096, .block_count=2048, .block_cycles=500, .cache_size=256, .lookahead_size=512, .read_buffer=2000226c, .prog_buffer=2000126c, .lookahead_buffer=2000026c, .name_max=0, .file_max=0, .attr_max=0})
lfs_trace:4066: lfs_fs_size(200033e8)
lfs_trace:3719: lfs_fs_traverse(200033e8, 25d, 20006240)

``

Gspan93, 2019-09-17:
I am facing similar error as @ksstms, but in my case, it's a non-empty chip with first two blocks erased & have never seen lfs.
Even if I try to format the chip first, without mounting it (following the example in Readme) I have a NULL cb in lf_fs_traverse().
Here's the trace log for reference:
    [CORTEX_M4_0] lfs_trace:3503: lfs_format(200033e8, ccf8 {.context=0, .read=9541, .prog=945d, 
    .erase=94ef, .sync=9459, .read_size=16, .prog_size=256, .block_size=4096, .block_count=2048, 
    .block_cycles=500, .cache_size=256, .lookahead_size=512, .read_buffer=2000226c, 
    .prog_buffer=2000126c, .lookahead_buffer=2000026c, .name_max=0, .file_max=0, .attr_max=0})
    lfs_trace:4066: lfs_fs_size(200033e8)
    lfs_trace:3719: lfs_fs_traverse(200033e8, 25d, 20006240)

ksstms, 2019-09-18:
@schoegmi thanks! Then I guess it's not the same problem as ours.

@geky I'm absolutely sure that it happened in `lfs_format`. Unfortunately I can not reproduce it now, because I have erased the chip and forgot to make a backup of the contents :angry: 
I think @Gspan93 might be having a similar issue