littlefs-project/littlefs #169: Test sanity of callbacks

meh, 2019-04-17:
Is there a way to verify the implementation for `read` and friends is sane and behaving the way littlefs expects it to?

I'm seeing some really weird behavior with littlefs and I'm pretty sure it's because either the NAND flash library provided by the manufacturer sucks or because my wrapper for that is doing something wrong, but I have no clue on how to verify either.

I read one of the issues talking about using littlefs on NAND and it seemed to work out nicely, so I'm pretty sure it's my fault, but still, any advice?

Thanks!

FreddieChopin, 2019-04-18:
If possible, you can always show the code for review here, as long as it's not 1000 lines long (; And also describe the symptoms you are seeing.

meh, 2019-04-18:
Problem with showing the code is I'm writing a C++ wrapper to work with the rest of a C++ frameworkish thing at work so it's kind of hard to show the debugging code I'm coming up with.

The behavior I'm seeing right now that is very puzzling is twofold:

1. It seems that if I have a loop that reads a file (open, read, close) then writes "foo" if the content is "bar" and "bar" if the content is "foo" (open, write, close) and then read the file (open, read, close) then the content doesn't change even if `lfs_file_write` is being called with the right buffer and no errors are reported.

2. Writing to a file in a loop (like a hundred `'A' .. 'Z'`) and then closing the file, then opening it results in no such file found, even if closing the file was successful.

This makes me think it's something very very wrong with my NAND flash interaction, but I don't even know where to start debugging that as I couldn't find documentation defining the behavior of the callbacks.

FreddieChopin, 2019-04-18:
Are you using v1 or v2? If v2 maybe it is related to some possible bugs in there, like #167 or something else? Did you make littlefs work with some other type of memory previously, or is this your first try?

Testing the low-level layer is probably best done with a debugger - call the "write" wrapper with a random (but valid) arguments, verify that it does what it is supposed to do, the call your "read" wrapper with the same arguments and compare the contents. Sorry - I don't have anything better.

meh, 2019-04-18:
Found the issue, one of the callbacks was returning the status value from the NAND flash library instead of something LFS understands, I guess that's what happens when the typesystem ain't strong enough.

Still, I think it would be valuable to have some sanity checking for callbacks, or at least a little more documentation on what they should return and how they should behave :panda_face:

meh, 2019-04-18:
@FreddieChopin the problem is it's not very clear what the callbacks are exactly supposed to do, but maybe I just missed some part of the documentation that outlines the expectations for the various callbacks.

geky, 2019-05-22:
Point taken, littlefs does need more documentation over the porting layer (related https://github.com/ARMmbed/littlefs/issues/56).

You're not the only person bitten by the bad handling of positive return values of block device drivers. Perhaps littlefs should handle these as "ok"?

I don't want to give the impression that littlefs can handle partial writes from the block device driver, but running off into the dirt and exploding is probably not the best way to handle it. Maybe we should add an assert on block device callbacks returning positive values?

meh, 2019-05-23:
Eh, honestly it's a case of type system being not strong enough, with C++ you could have an `enum class` and this wouldn't have happened.

I wouldn't silently be okay with things because if you're plainly returning whatever status code is coming from the underlying layer there's a chance stuff will break in unexpected ways.

What might somewhat help is debug asserts to check the return value is within expected values, but even then there's a chance it's just a coincidence that values match from the underlying layer.

So yeah, I think the only thing to do is thorough documentation for the porting layer, maybe using an `enum` for the return type of the callbacks and hope the compiler at least throws a warning there, and then hoping for the best.