littlefs-project/littlefs #170: file.block returns -2 when we have inline files

lsilvaalmeida, 2019-04-18:
Hello!

I'm new in all this FS context, but I'm trying to do a simple test and there is something strange. When I write files which have size enough to fill an entire block, there is no problem: file.block saves the value of the respective block. But when I write small files (2B each) and littlefs try to fit all the files in only one block, all the file.block save the value -2. Is this normal or am I doying something wrong?

Thank you in advance.

geky, 2019-05-22:
Hi @lsilvaalmeida, sorry for late response. This is normal, small files can be inlined into their parent's log. In this case we have a "block" associate with the file in the traditional sense, so a null value of -2 is used. The reason for -2 is because -1 is already used for indicating a flushed cache. These should probably be using an enum for clarity...

More info on inline files here:
https://github.com/ARMmbed/littlefs/blob/master/DESIGN.md#files

lsilvaalmeida, 2019-05-23:
Hmm, ok then. Thanks for you response. Can I see where the file will be writen to/ read from in the file struct? Or I have to know that it is a small file so I take the directory block?

geky, 2019-05-23:
Ah! So the lfs_file_t struct also contains a reference to its metadata pair. This is a two-block log that contains the metadata for the file. Note that it can contain metadata for multiple files.

``` c
file.m.pair[0]; // first block in metadata pair
file.m.pair[1]; // second block in metadata pair
file.ctz.head;  // first block in skip-list for file on disk
file.block;     // first block in skip-list for in-flight writes (writes that have not been committed yet)
```

I'm not exactly sure what you're trying to do, is this just for curiosity?

Something that may be useful is lfs_fs_traverse, which gives you every block that the filesystem is using:
https://github.com/ARMmbed/littlefs/blob/f35fb8c14866a4a4677756f6dbeca78f8a9b4001/lfs.h#L617-L624

lsilvaalmeida, 2019-05-24:
> I'm not exactly sure what you're trying to do, is this just for curiosity?

I want just see where a file will be writen to print on a debug log. But it's fine, I already deal with it if I see a block "-2" then I print "on the matadata block". 

> Something that may be useful is lfs_fs_traverse, which gives you every block that the filesystem is using:

Yes, I'm using the traverse function to see which blocks are in use and it's really helpfull. 

Thanks one more time for your time and response. Always helpfull.

closed 2019-05-24 by lsilvaalmeida