littlefs-project/littlefs #171: lfs_format reads blocks it never touched

meh, 2019-04-19:
`lfs_format` uses `lfs_dir_alloc` and `lfs_dir_commit` which read blocks, even if the data is ignored this should be avoided as it's unexpected behavior as far as I'm concerned (and prevents me from implementing an encrypted block device).

I was trying to fix this myself and send a pull request but I don't know enough about littlefs internals for this to not take me hours.

But yeah, in general reading erased blocks might break things even if the data is never used, so it would be nice if that could be avoided everywhere :panda_face: 

Thanks!

meh, 2019-04-22:
Also for the record, `lfs_format` actually uses the contents of the page it reads, it just assumes the content is `0xff`, `lfs_dir_alloc` uses it for `dir->rev`, just changing that part makes my encrypted block device thingamajig work.

geky, 2019-05-16:
Hi @meh, sorry for late response.

littlefs does read before erasing. This is a small optimization that saves an erase cycle when creating a metadata pair. I'll explain it below.

> it just assumes the content is 0xff

This is incorrect. It assumes the content is _consistent_ on successive reads, but the value can be arbitrary.

I've been able to get an encrypted block device to work by always returning the "decrypted" value of the block, even if it's garbage (as long as it's consistent garbage!).

If your decryption scheme is non-deterministic, I'd be very curious to know what you're up to :)

---

The reason for this read is to avoid an unnecessary erase when we create a metadata pair.

Metadata pairs are small, two-block logs that store a bunch of metadata updates. When we want to add a new metadata update but the block is full, we copy the metadata updates into the other block, removing any outdated updates. In order to tell which block is the most recent, we store a 32-bit revision count. The block with the highest revision count (using [serial comparison](https://en.wikipedia.org/wiki/Serial_number_arithmetic) to avoid issues with overflow) is the most recent.

```
                        commit B', need to compact
.----------------.----------------.    .----------------.----------------.
|   revision 1   |   revision 0   | => |   revision 1   |   revision 2   |
|----------------|----------------|    |----------------|----------------|
|       A        |                |    |       A        |       A'       |
|----------------|                |    |----------------|----------------|
|    checksum    |                |    |    checksum    |       B'       |
|----------------|                |    |----------------|----------------|
|       B        |                |    |       B        |    checksum    |
|----------------|                |    |----------------|----------------|
|       A'       |                |    |       A'       |       |        |
|----------------|                |    |----------------|       v        |
|    checksum    |                |    |    checksum    |                |
|----------------|                |    |----------------|                |
'----------------'----------------'    '----------------'----------------'
```

This is great, but when we create a new metadata pair it seems like we need to write both blocks with a revision count. Or do we?

Because the revision counts are cyclic and overflow, we can use any value as our initial revision count. So as long as data read is consistent, any garbage can be used to initialize our revision counts. We just mark the block with metadata as having a revision count greater than the garbage, and then littlefs will know to not use that block.

Erases are very expensive, so even though this only saves one erase on creation, it's worthwhile. And I have yet to see a block device where reads are _inconsistent_ before an erase.

More info here:
https://github.com/geky/littlefs/blob/master/DESIGN.md#metadata-pairs

meh, 2019-05-18:
The decryption result is deterministic, I should have written down what I was actually seeing happen instead of going all XY on you, sorry about that.

Right now I don't have access to my workstation (and I won't for the next few weeks) but I'll try to reproduce the issue locally, I just know it started working after making `lfs_format` happy by not reading on the first allocation, but considering my new understanding of metadata pairs the issue was just solved for the synthetic test-case I conjured up.

My test-case was reading a file, swapping content of the file depending on the content, then reading again and seeing what the content was, over and over, and that worked. I assume that doesn't create another metadata pair.

geky, 2019-05-22:
> I assume that doesn't create another metadata pair.

Ah, if you set `config.block_cycles`, then littlefs will move the metadata pair after that many erase cycles. This will call lfs_dir_alloc.

meh, 2019-05-23:
What happens when `config.block_cycles` is 0?

geky, 2019-05-23:
`config.block_cycles = 0` disables proactive wear leveling. In this case blocks are only moved when an error is detected.

Thinking about it, littlefs may still call `lfs_dir_alloc` if blocks develop errors. Unfortunately this doesn't happen until the block device nears end of life.

I would suggest setting `config.block_cycles` to something like 100 or 1000. Higher numbers give you faster performance at a tradeoff of less uniform wear-leveling. 0 is effectively infinity.

I've recently been wondering if defaulting to 0 has been a mistake. It seems easy to miss `config.block_cycles`.

meh, 2019-05-23:
Personally I used `block_cycles = 0` because the specs for the NAND flash I was using didn't really give me a clear number of cycles per block, so I assumed `0` would do _the right thing™_.

geky, 2019-05-23:
Yeah, I don't think it does. I'm thinking no wear-leveling should be changed to -1, and 0 should assert (or default to the _right thing™_)...

I'm not sure exactly what the best `block_cycles` is. It's an area I need to look into more and is likely dependent on the hardware. NAND flash has ~10,000 erase cycles from what I understand, but you would want to set `block_cycles` to something like 100 as this would let each block be used as a metadata pair ~100 times.

A word of warning, NAND support in littlefs is in its infancy. Sorry about the kinks that need ironing out.

meh, 2019-05-23:
I think `-1` for disable and `0` for the _right thing™_ would be best, I didn't realize `0` would disable wear leveling entirely truth be told :panda_face: 

And no worries about the kinks, I really like the focus on no-allocations (I'm building with `LFS_NO_MALLOC`) and low overhead and the design is a breathe of fresh air compared to FAT :+1:

geky, 2019-05-23:
Haha thanks

geky, 2020-01-27:
I believe this is fixed, feel free to reopen if I'm wrong.

closed 2020-01-27 by geky