littlefs-project/littlefs #172: Write truncates file after seek on v2

ebinans, 2019-04-24:
The following code shows how write truncates file after seek on v2:
```
lfs_file_open(&fs, &file, "why.bin", LFS_O_RDWR | LFS_O_CREAT);
lfs_file_write(&fs, &file, "12345678901234567890123456789", 0x1D);

printf("= File size before: %u\n", lfs_file_size(&fs, &file));

lfs_file_seek(&fs, &file, 0, LFS_SEEK_SET);
lfs_file_write(&fs, &file, "123456789012", 0xC);
lfs_file_sync(&fs, &file);

printf("= File size after:  %u\n", lfs_file_size(&fs, &file));
```

Output:
```
= File size before: 29
= File size after:  12
```

Latest v1 is not affected.

geky, 2019-05-28:
Thanks for creating an issue! I've pushed up a fix here: https://github.com/ARMmbed/littlefs/pull/194

Looks like the issue was an assignment to the wrong file structure field. Let me know if you see any issues with #194.

ebinans, 2019-05-30:
Great, thank you!

ebinans, 2019-06-05:
If we modify test case above to add a read then it fails with large positive value:

```
lfs_file_open(&fs, &file, "why.bin", LFS_O_RDWR | LFS_O_CREAT);
lfs_file_write(&fs, &file, "12345678901234567890123456789", 0x1D);

printf("= File size before: %u\n", lfs_file_size(&fs, &file));

lfs_file_seek(&fs, &file, 0, LFS_SEEK_SET);
lfs_file_write(&fs, &file, "123456789012", 0xC);

printf("= File size after:  %u\n", lfs_file_size(&fs, &file));

char buffer[5];

lfs_file_seek(&fs, &file, lfs_file_size(&fs, &file) - sizeof(buffer), LFS_SEEK_SET);
lfs_ssize_t rd = lfs_file_read(&fs, &file, buffer, sizeof(buffer));

printf("= Expect: %d, Got: %d\n", sizeof(buffer), rd);
```

We get the following:
```
* Formatted and mounted
= File size before: 29
= File size after:  29
= Expect: 5, Got: 537920512
```

It appears write is fixed but read unexpectedly fails with large positive number.

geky, 2019-06-10:
Good find! Looks like this was a similar issue. Except instead of not updating the file's metadata it was the cache itself that needed to be updated correctly.

I added a commit to #194 that should fix this.

And here I was thinking I had decent testing over seek. Hopefully that's the last issue you run into.

ebinans, 2019-06-11:
Seems to be good now, thank you!

closed 2019-06-11 by ebinans