littlefs-project/littlefs #173: Its difficult to not using lfs_malloc in V2.0.0

ShareCat, 2019-04-26:
Its difficult to not using lfs_malloc in V2.0.0, not all the MCU support malloc functions.

Since this code: static const struct lfs_file_config defaults = {0};

Its difficult to initial the file->cfg->buffer at the first beginning of main function.

jimparis, 2019-05-16:
If you need to initialize `file->cfg->buffer`, just use `lfs_file_opencfg` instead of `lfs_file_open`.

geky, 2019-05-16:
Sorry it did get a bit more difficult.

I'd rather keep it at one way of doing things, and the per-file buffer (with lfs_file_opencfg) is more powerful. (There's also a code cost for each config option).

Alternatively you could provide a custom lfs_malloc/lfs_free that provides the file buffer.

``` c
#define CACHE_SIZE 512
static uint8_t file_buffer[CACHE_SIZE];

void *lfs_malloc(size_t size) {
    return file_buffer;
}

void lfs_free(void *p) {
    (void)p;
}
```

To do this you would need to either modify lfs_util.h or provide your own lfs_config.h by compiling with -DLFS_CONFIG=lfs_config.h.

---

This could be extended to provide memory for a fixed number of buffers with a fixed-size allocator. Fixed-size allocators are much cheaper than fully dynamic memory allocation.

``` c
#define CACHE_SIZE 512
#define FILE_COUNT 4
static uint8_t file_buffer[FILE_COUNT * CACHE_SIZE];
static void *file_buffer_list;
static bool file_buffer_initialized = false;

void *lfs_malloc(size_t size) {
    if (!initialized) {
        // create linked list
        for (int i = 0; i < FILE_COUNT-1; i++) {
            *(void**)file_buffer[i * CACHE_SIZE] = &file_buffer[(i+1) * CACHE_SIZE];
        }
        file_buffer_list = &file_buffer[0];
        *(void**)file_buffer[(FILE_COUNT-1) * CACHE_SIZE] = NULL;
        initialized = true;
    }

    if (!file_buffer_list) {
        return NULL;
    }

    // grab first block from free list
    void *p = file_buffer_list;
    file_buffer_list = *(void**)p;
    return p;
}

void lfs_free(void *p) {
    if (!p) {
        return;
    }

    // prepend to free list
    *(void**)p = file_buffer_list;
    file_buffer_list = p;
}
```