littlefs-project/littlefs #176: Mount two littlefs on a single memory chip

Artem729, 2019-05-04:
I am working on a project in which I need to use two file systems. Is it possible to mount two littlefs on a single memory chip, (I use winbond 25q64)? The first file system occupies the first seven megabytes of memory, the second occupies the remaining one megabyte. If this is possible, please tell me exactly how this can be done.

e107steved, 2019-05-07:
Yes, its perfectly possible.
You need a separate set of low-level entry points for each file system; for the second set of routines, just add a 7 megabyte offset to the addresses you use.

geky, 2019-05-22:
To build on @e107steved's answer, you can mount multiple littlefs instances by providing different `struct lfs_config` structs. This is where you provide the block device operations, so all you need is one block device with the offset and one without.

closed 2019-08-19 by Artem729