littlefs-project/littlefs #178: v2.0.3. Got LFS_IS_DIR while opening previously created file

Roger-Waters, 2019-05-13:
```
fd = f_open(IQ_BOOT_COUNT_FILEPATH, O_RDWR | O_CREAT);
  if(fd < 0) {
    LOG_WARN("fopen %d", fd);
  }

  ssize = f_read(fd, &boot_count, sizeof(boot_count));
  if(ssize < 0) {
    LOG_WARN("fread %d", ssize);
  }

  boot_count += 1;
  ssize = f_rewind(fd);
  if(ssize != LFS_ERR_OK) {
    LOG_WARN("frewind %d", ssize);
  }

  ssize = f_write(fd, &boot_count, sizeof(boot_count));
  if(ssize < 0) {
    LOG_WARN("fwrite %d", ssize);
  }

  // remember the storage is not updated until the file is closed successfully
  fd = f_close(fd);
  if(fd != LFS_ERR_OK) {
    LOG_WARN("fclose %d", fd);
  }
```
f_* functions is just a wrappers for lfs_*. It uses array of statically allocated lfs_file_t pointers and call alloc() inside.

Config looks like this
```
  lfs_cfg[INTERNAL_PARTITION].read            = internal_eeprom_read;
  lfs_cfg[INTERNAL_PARTITION].prog            = internal_eeprom_prog;
  lfs_cfg[INTERNAL_PARTITION].erase           = internal_eeprom_erase;
  lfs_cfg[INTERNAL_PARTITION].sync            = internal_eeprom_sync;
  lfs_cfg[INTERNAL_PARTITION].read_size       = 128UL;
  lfs_cfg[INTERNAL_PARTITION].prog_size       = 128UL;
  lfs_cfg[INTERNAL_PARTITION].cache_size      = 128UL;
  lfs_cfg[INTERNAL_PARTITION].lookahead_size  = 16UL * 8;
  lfs_cfg[INTERNAL_PARTITION].block_size      = FileSpaceStandard;
  lfs_cfg[INTERNAL_PARTITION].block_count     = defFsNumberOfDefFiles;
  lfs_cfg[INTERNAL_PARTITION].context         = (void *)0; // offset
  lfs_cfg[INTERNAL_PARTITION].block_cycles    = 300;
  lfs_cfg[INTERNAL_PARTITION].name_max        = 0;
  lfs_cfg[INTERNAL_PARTITION].file_max        = 0;
  lfs_cfg[INTERNAL_PARTITION].attr_max        = 0;
```
Any another info needed?

**UPD**: My small investigation gets next results:

* On first boot, when `boot_count` does not exist `lfs_file_open()` function works as expected.
* On second boot, when `boot_count` has already created, `lfs_file_open()` returns LFS_ERR_ISDIR (-21) error.

I tried to comment out corresponding checks in `lfs_file_opencfg()` implementation and had no error now. This hunk of code:
```
    } else if (flags & LFS_O_EXCL) {
        err = LFS_ERR_EXIST;
        goto cleanup;
    /* NEXT THREE LINES */
    // } else if (lfs_tag_type3(tag) != LFS_TYPE_REG) {
        // err = LFS_ERR_ISDIR;
        // goto cleanup;
    } else if (flags & LFS_O_TRUNC) {
        // truncate if requested
        tag = LFS_MKTAG(LFS_TYPE_INLINESTRUCT, file->id, 0);
        file->flags |= LFS_F_DIRTY;

```

So, I think, error is around here.

**UPD2**. Actually, `lfs_tag_type3()` for this file returns `0x0000`

geky, 2019-05-22:
Hi @Roger-Waters, that's interesting. A type3 of 0x000 should never exist, something must being going wrong.

What is your block size (the FileSpaceStandard)? Do you have asserts enabled?

What does your `struct lfs_file_config` struct look like for `lfs_file_opencfg`? Are you using malloc for the filesystem caches?

sipke, 2019-06-18:
I'm facing the same issue doing a port of littlefs, and removing said 3 lines hides the problem.

I have a static buffer passed to .buffer of the struct lfs_file_config given lfs_file_opencfg
```
struct lfs_file_config fconfig;
...
memset(&fconfig, 0, sizeof(struct lfs_file_config));
fconfig.buffer = allign_64bit(fileBuffer);
...
err = lfs_file_opencfg(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT, &fconfig);
```
not sure the 64bit allignment is necessary, but did so just in case.
I had block_count of 128 (and tried 0x8000) and block_size of 4096 and 16 for read, prog, lookahead and cache sizes. All buffers static.

I've tried to step through the code to try and locate where exactly the tag is written, but have not figure it out yet. If you could point me to where the LFS_TYPE_REG is  written I can continue looking at where the issue is. Cheers.

Roger-Waters, 2019-07-30:
Hi again. Recently I tried to upgrade to 2.1.0 and got the same error. So, this is my bunch of code:

```
#define INTERNAL_PARTITION              0
#define LFS_MAX_PARTITION               2
  /* ... */
  lfs_t              fs_partitions[LFS_MAX_PARTITION];
  struct lfs_config  lfs_cfg[LFS_MAX_PARTITION];

  lfs_cfg[INTERNAL_PARTITION].read            = internal_eeprom_read;
  lfs_cfg[INTERNAL_PARTITION].prog            = internal_eeprom_prog;
  lfs_cfg[INTERNAL_PARTITION].erase           = internal_eeprom_erase;
  lfs_cfg[INTERNAL_PARTITION].sync            = internal_eeprom_sync;
  lfs_cfg[INTERNAL_PARTITION].read_size       = 64UL;
  lfs_cfg[INTERNAL_PARTITION].prog_size       = 64UL;
  lfs_cfg[INTERNAL_PARTITION].cache_size      = 64UL;
  lfs_cfg[INTERNAL_PARTITION].lookahead_size  = 16UL;
  lfs_cfg[INTERNAL_PARTITION].block_size      = 128UL;
  lfs_cfg[INTERNAL_PARTITION].block_count     = 31UL;
  lfs_cfg[INTERNAL_PARTITION].context         = (void *)0; // offset
  lfs_cfg[INTERNAL_PARTITION].block_cycles    = 300;
  lfs_cfg[INTERNAL_PARTITION].name_max        = 0;
  lfs_cfg[INTERNAL_PARTITION].file_max        = 0;
  lfs_cfg[INTERNAL_PARTITION].attr_max        = 0;
  /* ... */
  lfs_mount(&fs_partitions[INTERNAL_PARTITION], &lfs_cfg[INTERNAL_PARTITION]);
```
Yes, I use malloc function implemented by myself. It uses statically allocated memory pool.
Some code from confg_lfs.h
```
/* ... */
static inline void *lfs_malloc(size_t size) {
#ifndef LFS_NO_MALLOC
    return membag_alloc(size);
#else
    return NULL;
#endif
}

// Deallocate memory, only used if buffers are not provided to littlefs
static inline void lfs_free(void *p) {
#ifndef LFS_NO_MALLOC
    membag_free(p);
#endif
}
/* ... */
```

Roger-Waters, 2020-06-13:
Hi everyone. I'm just trying to upgrade my project again to last version 2.2 and "boot_count example" is finally working. It looks like #299 fixes an issue. Still in progress of full porting LFS 2.2 to Atmega2560.