littlefs-project/littlefs #180: v2.0.3 seems to corrupt existing files when writing many small files in one directory

Mohawi, 2019-05-16:
We tested littleFS v2.0.3 with a blockdevice that offers 20 erase blocks á 256 bytes and a program / read blocksize of 32 byte. The blockdevice does not simulate failing blocks. So every read/write access is executed correctly.
The wearlevelling cycle count is set to 5 cycles.

The following code creates 50 files (which should be stored inline) in an directory and always writes the same data to them. Afterwards the files are read back and compared to originally written data, which should not produce any errors. 
So far so good. The program calls do not return any error. Trying to open the files, we get an error "LFS_ERR_NOENT".
```
...
char aWriteBuffer[2];
char aReadBuffer[sizeof(aWriteBuffer)];

for(uint32_t i=0; i<50; i++)
{
  lfs_file_t FileHandle ={0};
  std::stringstream  sFilename;
  sFilename << "Home/" << i;
  int32_t i32Error = 0;
  i32Error = lfs_file_open(&m_LittleFS, &FileHandle, sFilename.str().c_str(), LFS_O_CREAT | LFS_O_WRONLY);
  if(i32Error < 0) 
  {
    std::cout << "Error 1:" << i32Error << std::endl;
  }
  else
  {
    memset(aWriteBuffer, i, sizeof(aWriteBuffer));
    i32Error = lfs_file_write(&m_LittleFS, &FileHandle, aWriteBuffer, sizeof(aWriteBuffer));
    if(i32Error < 0) std::cout << "Error 2:" << i32Error << std::endl;
    i32Error = lfs_file_close(&m_LittleFS, &FileHandle);
    if(i32Error < 0) std::cout << "Error 3:" << i32Error << std::endl;
  }
}
std::cout << "-----------------READ----------------------" << std::endl;
for(uint32_t i=0; i<50; i++)
{
  lfs_file_t FileHandle ={0};
  std::stringstream  sFilename;
  sFilename << "Home/" << i;
  int32_t i32Error = 0;
  memset(aWriteBuffer, i, sizeof(aWriteBuffer));
  i32Error = lfs_file_open(&m_LittleFS, &FileHandle, sFilename.str().c_str(), LFS_O_RDONLY);
  if(i32Error < 0) 
  {
    std::cout << "Error 4:" << i32Error << std::endl;
  }
  else
  { 
    i32Error = lfs_file_read(&m_LittleFS, &FileHandle, aReadBuffer, sizeof(aReadBuffer));
    if(i32Error < 0) std::cout << "Error 5:" << i32Error << std::endl;
    i32Error = lfs_file_close(&m_LittleFS, &FileHandle);
    if(i32Error < 0) std::cout << "Error 6:" << i32Error << std::endl;
    if(memcmp(aWriteBuffer, aReadBuffer, sizeof(aReadBuffer)) == 0) std::cout << i << " OK" << std::endl;
  }
}
```
The result:
```
-----------------READ----------------------
Error4:-2
Error4:-2
Error4:-2
Error4:-2
4 OK
5 OK
6 OK
7 OK
8 OK
9 OK
Error4:-2
Error4:-2
...
```

Is there something that we did wrong, is there a maximal amount of files per directory or is this simply a bug in littelFS?

bigbrett, 2021-03-22:
@Mohawi did you ever find a solution? I'm seeing `lfs_open` return `-2` on certain devices as well and can't figure out why. @geky do you know how this would be possible if the file names never change?  and they are always being opened with the same mode?

geky, 2021-03-23:
Sorry, I don't have an answer. I'm not aware of a known bug, though I haven't been able to investigate @Mohawi's original issue. LittleFS is also at v2.4.0 now (>v2.0.3), so it's possible this was fixed in an unrelated patch.

Something to note is that unlike FAT-like filesystems, LittleFS rewrite's a file's metadata as a part of wear-leveling. So even if you're not changing the file name there's still a lot of space for something to go wrong.

You could try reproducing the issue in a test script (I really should document this) with your device's geometry:
https://github.com/littlefs-project/littlefs/blob/master/tests/test_files.toml#L4

That would narrow it down to if it's a driver issue or filesystem issue. If you can get a test script that fails that would be a really helpful starting point for debugging the issue.

bigbrett, 2021-03-29:
Thanks for getting back to me @geky, I appreciate your time. 

I'm actually seeing this error with LittleFS v2.3.0. Would you like me to make a new issue and move discussion there? 

I don't think I'd be able to make a test script for it, since it actually works fine up until it when it doesn't. After some unknown number of file operations, all of a sudden `lfs_open` just returns -2. I think you already have a test script that covers our case, as all we do is: 

- POR, format and mount if successful
- about 12 open --> read --> close on different files to load initial configuration parameters
- a bunch of event-driven open-->read-->close and open-->write-->close sequences if parameters are to be updated during runtime
- power off whenever

We have devices in the field (robotics application) that are locking up, and when a debugger is attached, it is because we are asserting that `lfs_open() == LFS_ERR_OKAY` for every parameter when loaded after a POR. Somewhere along the line, after a power cycle, `lfs_open` starts returning -2. But there is a large period of time before that when it works fine. 

I tried to debug further  by downloading the region of flash memory to a file on my host machine via SEGGER JLink, but am unable to mount it via `littlefs-fuse`...this is probably user error though. When I try and mount the loopback device on my host machine to inspect the filesystem, I get:

```
brett: $ sudo chmod a+rw /dev/loop14                                                                                     
brett: $ losetup /dev/loop14 ../lfs-bug-again.mem
  losetup: ../lfs-bug-again.mem: Warning: file does not fit into a 512-byte sector; the end of the file will be ignored.
brett: $ ./lfs /dev/loop14 mount
  littlefs/lfs.c:1070:error: Corrupted dir pair at {0x1, 0x0}
  lfs_fuse.c:514:error: Invalid or incomplete multibyte or wide characte
It mounts fine on the target device....might need to fiddle with default sector/block sizes somehow?

Any and all help greatly appreciated. **I'd be happy to get you the raw filesystem binary that I pulled from the target device, if you would be willing to look at it.** Maybe it is obvious if you know what to look for `¯\_(ツ)_/¯`

bigbrett, 2021-03-29:
you can disregard the fuse issues, I solved by fixing block size. However, once I mount, sure enough, the file is not there! So something is causing it to be "erased" (more likely that metadata is being overwritten?). There are still references to the original filename in the binary though, it just doesn't show up when mounted (explaining the -2). Any suggestions on tests to run that would shed light on where exactly the file is getting corrupted? I never have anywhere in my code that removes the file, so something nefarious is going on....