littlefs-project/littlefs #182: v1 -> v2 migration doesn't work

rojer, 2019-05-16:
lfs_migrate succeeds (returns 0) but the filesystem cannot be mounted as v2 (corrupted dir pair).

i'm using the following simple program - https://gist.github.com/rojer/3b48a8630c7907f745901eb28b86265b

build and run:
```
$ cc -Wall -Werror -DLFS_MIGRATE -I./littlefs -I./littlefs1 -o test ./littlefs/lfs.c ./littlefs/lfs_util.c ./littlefs1/lfs1.c ./littlefs1/lfs1_util.c test.c && ./test
```

`./littlefs` is https://github.com/ARMmbed/littlefs/commit/f35fb8c14866a4a4677756f6dbeca78f8a9b4001 + https://github.com/ARMmbed/littlefs/pull/181
`./littlefs1` is https://github.com/ARMmbed/littlefs/commit/4827795e4b8b4b1c849ac50347fb49c4dd443964 (latest v1-prefix branch).

first it formats a memory-backed lfs1, writes a single file called "test.txt" and unmounts it:
```
R   #0000 64 @ 0+0   
R   #0001 64 @ 2+0                                                     
  E #0000 4096 @ 3
 W  #0000 64 @ 3+0
R   #0002 64 @ 3+0
  E #0001 4096 @ 1
 W  #0001 64 @ 1+0
R   #0003 64 @ 1+0
  E #0002 4096 @ 0
 W  #0002 64 @ 0+0
R   #0004 64 @ 0+0
R   #0005 64 @ 1+0
lfs1_format: 0
R   #0006 64 @ 0+0
R   #0007 64 @ 1+0
R   #0008 64 @ 0+0
R   #0009 64 @ 3+0
R   #0010 64 @ 2+0
R   #0011 64 @ 0+0
R   #0012 64 @ 1+0
R   #0013 64 @ 0+0
R   #0014 64 @ 3+0
R   #0015 64 @ 2+0
  E #0003 4096 @ 2
 W  #0003 64 @ 2+0
R   #0016 64 @ 2+0
lfs1_file_open: 0
  E #0004 4096 @ 4
lfs1_file_write: 4
 W  #0004 64 @ 4+0
R   #0017 64 @ 4+0
R   #0018 64 @ 2+0
R   #0019 64 @ 3+0
R   #0020 64 @ 2+0
  E #0005 4096 @ 3
 W  #0005 64 @ 3+0
R   #0021 64 @ 3+0
lfs1_file_close: 0
lfs1_unmount: 0
```

it then tries to v2-mount it, which understandably fails:
```
R   #0000 64 @ 0+0
R   #0001 64 @ 1+0
R   #0002 256 @ 0+0
R   #0003 256 @ 1+0
lfs error:966: Corrupted dir pair at 0 1
lfs_mount: -84
```
then it the migrates the fs in-place, which finishes successfully:
```
R   #0000 64 @ 0+0
R   #0001 64 @ 1+0
R   #0002 64 @ 0+0
R   #0003 64 @ 3+0
R   #0004 64 @ 2+0
R   #0005 64 @ 0+0
R   #0006 64 @ 1+0
R   #0007 64 @ 0+0
R   #0008 64 @ 1+0
R   #0009 256 @ 0+0
R   #0010 256 @ 1+0
lfs error:966: Corrupted dir pair at 0 1
R   #0011 64 @ 3+0
R   #0012 64 @ 2+0
R   #0013 64 @ 3+0
R   #0014 64 @ 2+0
R   #0015 256 @ 3+0
R   #0016 256 @ 2+0
lfs error:966: Corrupted dir pair at 3 2
R   #0017 64 @ 6+0
  E #0000 4096 @ 5
R   #0018 128 @ 5+64
 W  #0000 64 @ 5+0
R   #0019 64 @ 5+0
R   #0020 64 @ 3+0
R   #0021 64 @ 6+0
R   #0022 64 @ 5+0
R   #0023 256 @ 5+64
R   #0024 64 @ 6+0
R   #0025 64 @ 5+0
R   #0026 256 @ 5+64
 W  #0001 64 @ 5+64
R   #0027 128 @ 5+64
lfs debug:4370: Migrating 6 5 -> 3 2
  E #0001 4096 @ 2
R   #0028 64 @ 6+0
R   #0029 64 @ 5+0
R   #0030 256 @ 5+64
R   #0031 128 @ 5+0
R   #0032 64 @ 0+0
R   #0033 64 @ 1+0
  E #0002 4096 @ 1
R   #0034 128 @ 1+64
 W  #0002 64 @ 1+0
R   #0035 64 @ 1+0
R   #0036 64 @ 0+0
R   #0037 64 @ 1+0
R   #0038 256 @ 1+64
lfs_migrate: 0
```

finally it tries to v2-mount the fs again, and it still fails with corrupt dir pair, albeit with more i/o.
```
R   #0000 64 @ 0+0
R   #0001 64 @ 1+0
R   #0002 256 @ 1+64
R   #0003 64 @ 1+0
R   #0004 64 @ 3+0
R   #0005 64 @ 2+0
R   #0006 256 @ 3+0
R   #0007 256 @ 2+0
lfs error:966: Corrupted dir pair at 3 2
lfs_mount: -84
```

geky, 2019-05-28:
Sorry about the delay on this one. I just put up a fix here: https://github.com/ARMmbed/littlefs/pull/193

A quick note it looks like you are missing a call to lfs1_mount after lfs1_format, I'm surprised it works as is. Though I was able to reproduce the bug, so many thanks for the example program.

It looks like I simply missed flushing the cache at the end, which would make lfs_migrate think it succeeded but not leave the filesystem in a valid state. Unfortunately the tests I did all ended up cache-aligned which causes the caches to auto-flush.

Let me know if the migration works with this fix. Thanks for creating an issue!

rojer, 2019-05-28:
indeed, that fixes the issue. thanks!

i have one more question re: migration. it looks like after migration one of the original superblock pair, namely block 0, is still a valid v1 superblock and thus v2 superblock is not redundant. should the v2 superblock be written in both blocks?

rojer, 2019-05-28:
this is the fs image after migration:
```
$ hexdump -C fs.img 
00000000  03 00 00 00 34 00 00 00  03 00 00 00 02 00 00 00  |....4...........|
00000010  2e 14 00 08 03 00 00 00  02 00 00 00 00 10 00 00  |................|
00000020  40 00 00 00 01 00 01 00  6c 69 74 74 6c 65 66 73  |@.......littlefs|
00000030  0a 38 cf 0b ff ff ff ff  ff ff ff ff ff ff ff ff  |.8..............|
00000040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
00001000  04 00 00 00 f0 0f ff f7  6c 69 74 74 6c 65 66 73  |........littlefs|
00001010  2f e0 00 10 00 00 02 00  00 10 00 00 40 00 00 00  |/...........@...|
00001020  ff 00 00 00 ff ff ff 7f  fe 03 00 00 40 0f fc 10  |............@...|
00001030  03 00 00 00 02 00 00 00  30 10 00 0c fe aa ae 1c  |........0.......|
00001040  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
*
```

geky, 2019-05-28:
Hmm, only one of the pair is strictly necessary. For most metadata operations littlefs leaves the old data in the backup block as otherwise writing metadata would cost twice as many erases.

Actually, at this point, both v2 and v1 exist in a completely valid state on the disk, sharing as much COW data as is possible. This isn't necessarily a good thing as v1 will fall out of sync as v2 begins to cannibalize it. If you try to mount v1 after writing to v2, it won't necessarily detected an error even if it's been corrupted. This caused Mbed OS some pain before when users were switching between FAT and littlefs all the time.

On the other hand, the reason for keeping v1 intact in the first place is so that if the migration fails you can fall back to v1 which will still be safe to use.

rojer, 2019-05-28:
> This isn't necessarily a good thing as v1 will fall out of sync as v2 begins to cannibalize it. If you try to mount v1 after writing to v2, it won't necessarily detected an error even if it's been corrupted. 

yes, it's not a good thing. apart from right after migration, the state of v1 tree is unknown and thus not really useful. i think there's a strong argument in favor of putting the final nail into v1 by erasing its superblock at the end of (an otherwise successful) migration process.

geky, 2019-05-28:
Sounds reasonable to me. I'm going to need to figure out the best way to force a compact internally (simply erasing may be a noop).

geky, 2019-05-29:
Figured it out, added to #193

closed 2019-07-01 by geky