littlefs-project/littlefs #186: How to get free, useable remaining space?

Mohawi, 2019-05-17:
Hi, I'm wondering how to get the remaining (estimated) free, useable space of the configured block count of littleFS.
I can get the currently used space by calling `lfs_fs_traverse`. But how can I get the amount of already worn out blocks? Is there a neat way to do it?

sshaw-lutron, 2019-05-17:
I don't believe that littleFS actually keeps track of warn out blocks, it just moves to a new block when it detects a write failure. I suspect there _isn't_ a neat way of figuring out the number of bad blocks.
 
Also, `lfs_fs_size` returns the number of allocated blocks (basically does the `lfs_fs_traverse` calls for you). Might be a little easier to use ;-)

Mohawi, 2019-05-17:
Mh, ok. I see.

Thanks for pointing out the `lfs_fs_size`.

closed 2019-05-17 by Mohawi