littlefs-project/littlefs #187: v2.0.3 crashes when superblock becomes unwriteable

Mohawi, 2019-05-17:
When testing the littleFS with small files we mocked a blockdevice in a way that it simulates a failing flash memory after some write cycles.
Creating files causes writes to the superblock which then fails after some file writes. 

With v2.0.3 the superblock does not get relocated. Once it fails, the whole filesystem crashes.

Currently we use a workaround for this matter and create a "home" directory where all other file / directory actions take place in.

sshaw-lutron, 2019-05-17:
Based on my reading of Spec.md "The filesystem starts with only the root directory. The superblock metadata pairs grow every time the root pair is compacted in order to prolong the life of the device exponentially." It seems like the superblock should be expanding the mitigate the risk of wear out. Are you not seeing this?

I was planning on not using root directory to store files based on that text from Spec.md so that the superblock does not expand and use more space than necessary. Perhaps that is a misunderstanding on my part. I have not actually tested this aspect yet...

geky, 2019-05-22:
Sorry for late response, working through a backlog.

I hadn't thought of this case... Requires a bit of explanation.

Short answer is to set `config.block_cycles` to something on the order of 100 or 1000. This enables proper dynamic wear leveling. Higher numbers give you better performance, but less uniform wear.

---

Explanation:

v2 introduces several things around wear-leveling, one was the `config.block_cycles` configuration option to evict metadata pairs early instead of waiting for the block device to report errors. This introduces true dynamic wear-leveling which is important when you can't rely on detecting errors.

Setting `config.block_cycles` to 0 disables dynamic wear-leveling.

In theory this should be fine, littlefs will still behave as it did in v1 and respond to block device errors.

Except what you've found here is the result of this mixed with a different change.

As @sshaw-lutron pointed out, v2 also introduced expanding superblocks. That is the idea that we increase the amount of indirection in the superblock if we notice the superblock is being written to often.

This was a small change required for block devices > 1 GiB because the single level of indirection in v1 wouldn't be enough to avoid the superblock failing early at that scale.

```
 .--------.  .--------.  .--------.  .--------.  .--------.
.| super  |->| super  |->| super  |->| super  |->| file B |
|| block  | || block  | || block  | || block  | || file C |
||        | ||        | ||        | || file A | || file D |
|'--------' |'--------' |'--------' |'--------' |'--------'
'--------'  '--------'  '--------'  '--------'  '--------'

\----------------+----------------/ \----------+----------/
          superblock pairs               root directory
```

But on the other side, this change lets us actually shrink the superblock. Because we are dynamically increasing the indirection of the superblock, we can collapse the root block + superblock together unless we notice they are written often. This avoids the 4 block cost for littlefs that was in v1.

```
 .--------.  .--------.
.| super  |->| file B |
|| block  | || file C |
|| file A | || file D |
|'--------' |'--------'
'--------'  '--------'

\----+----/
 superblock pair
\----------+----------/
     root directory
```

**But**, how do we know the superblock is being written to? I just ended up using the `block_cycles` variable here. Unfortunately this means if `block_cycles` is 0, the default, the superblock will never get expanded.

From seeing how people are using v2, I'm wondering if letting `block_cycles == 0` disable wear leveling was a mistake and if instead `block_cycles == 0` should assert. It's easy to miss.