littlefs-project/littlefs #188: File append is not working after n cycles or 2000 bytes of data in NAND flash

joicetm, 2019-05-18:
Hi,

I am using lfs with NAND flash w25m01gv and all basic functions seems to be working fine. i was able to read and write multiple files with size up to 10000 bytes.
While trying to append the file with my data is getting corrupted with zeros, ie .. only last few writes are preserved. rest of the data comes as zeros.

Also i have observed that it happens when lfs moves the data to a new block. i have also tested the same with a NOR flash and that does not have any append problems it works fine. 

here is my configuration struct,

    .read_size = 4096,
    .prog_size = 4096,
    .block_size = 2048*64,
    .block_count = 128,
    .cache_size = 4096,
    .lookahead_size = 2048,

Has any body here faced similar issues?

geky, 2019-07-01:
Hi @joicetm, sorry for taking so long to get to this issue. There was a fundamental issue in how littlefs handles large (>1024) prog_sizes. NAND support is new for littlefs, sorry if it's not all fleshed out at the moment.

This issue is related and has more info: #195 

I pushed up a fix here: https://github.com/ARMmbed/littlefs/commit/7051bf58182e47dae115a3148b95825fc6c234a1, does this fix the issue you're seeing?

That fix is more a workaround than anything and it should be possible to improve the performance (https://github.com/ARMmbed/littlefs/issues/195#issuecomment-507458216), but hopefully it at least gets you all unblocked.

joicetm, 2019-07-02:
Thanks for the update, Geky.
I will check it, currently i have used a workaround by discarding the 1st 2k.