littlefs-project/littlefs #189: Wearleveling

friguimahdi, 2019-05-20:
Hello,

friguimahdi, 2019-05-20:
Hello , is there anyway to detect littlefs wearleveling managament ? how can i know if it's working properly or not ? 

Thank you for your response.

lsilvaalmeida, 2019-05-21:
Hello @friguimahdi 

I don't know if there is a better way to do it, but I store an erase count for each block in the block device structure. Then I can see how much a block has been erased over time. 

Here my device structure

typedef struct flash {
	/*partition device*/
	UINT16 dev;
    struct {
		UINT32 read_count;
		UINT32 prog_count;
		UINT32 erase_count[LFS_BLOCK_COUNT];
    } stats;
    /* to know if we are coming from a warm reset */
    UINT8 magic_number;
} flash_t;

friguimahdi, 2019-05-22:
@lsilvaalmeida Thank you for your reponse. 
I still don't get how can i track the wearleveling mechanism when i know how many times every block has been erased ? 
One thing i got in mind, is that the erase count of every block have to be the same if the wearleveling is activated ?

geky, 2019-05-22:
Hi @friguimahdi 

> One thing i got in mind, is that the erase count of every block have to be the same if the wearleveling is activated ?

Note: to activate wear leveling set `config.block_cycles` to something on the order of 100 or 1000. Higher numbers give you better performance, but less uniform wear.

The erase count of every block won't the exact same. It approaches a uniform distribution, but is not perfect.

[image]

This is a tradeoff for performance/code size.

---

One method I've seen for testing wear leveling:
1. Create simulated block device with size n
2. Run filesystem operations until simulated block devices dies
3. Create simulated block device with size 2n
4. Run filesystem operations until simulated block devices dies

Wear leveling is working if the larger block device runs for roughly 2x the number of filesystem operations.

To make this work you will need to have a simulated block device that simulated death. One option is returning `LFS_ERR_CORRUPT`, which littlefs will treat as a corrupt block, or a more realistic option would be to flip bits on the block device based on how erased the block is.

This is used in Mbed OS, but the code is not easy to follow:
https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/TESTS/filesystem_recovery/wear_leveling/main.cpp