littlefs-project/littlefs #190: If LFS_ATTR_MAX < cache_size, lfs does not make inline files

lsilvaalmeida, 2019-05-21:
Hello,
I'm working with LFS and it is really awsome! But, when I tried to implement it in my embeeded system, I couldn't make inline files, even when my files are smaller than the cache and than 1/4 of block_size. 
So I dug into the code and I've see that, in the `lfs_file_write` function, when we check if our file can be write in the same block of its directory, there is a comparision with LFS_ATTR_MAX, like we can see here:

```
if ((file->flags & LFS_F_INLINE) &&
            lfs_max(file->pos+nsize, file->ctz.size) >
            lfs_min(LFS_ATTR_MAX, lfs_min(
                lfs->cfg->cache_size, lfs->cfg->block_size/8))) {
        // inline file doesn't fit anymore
        file->off = file->pos;
        lfs_alloc_ack(lfs);
        int err = lfs_file_relocate(lfs, file);
        if (err) {
            file->flags |= LFS_F_ERRED;
            return err;
        }
    }
```

 I'm not using custom attibutes, so I've defined it as zero. Then ` lfs_min(LFS_ATTR_MAX, lfs_min(lfs->cfg->cache_size, lfs->cfg->block_size/8))` is surely 0 and we go inside the context where we don't have enough space for the file in this block and we call relocate.

I've noticed the comment `// no real benefit to using a smaller LFS_ATTR_MAX. Limited to <= 1022.`, but wasn't clear for me this consequence.

If this is the expected behavior, let me know. 

Regards

geky, 2019-05-22:
Hi @lsilvaalmeida, thanks : )

And good find! Thanks for creating an issue.

You're right this shouldn't be LFS_ATTR_MAX, LFS_ATTR_MAX is unrelated to inline files, though at one point during development they were related.

It should actually be limited by the bits available in the tag's size field, I pushed up a patch here and can merge it unless you see any issues: https://github.com/ARMmbed/littlefs/commit/0c363d26660d9f985d109793948fa2b56231478f

---

> I've noticed the comment // no real benefit to using a smaller LFS_ATTR_MAX. Limited to <= 1022., but wasn't clear for me this consequence.

I've considered dropping LFS_ATTR_MAX. It's only there for consistency with the other limits (LFS_FILE_MAX and LFS_NAME_MAX).

LFS_NAME_MAX is useful because the lfs_info struct has a buffer containing the name. Decreasing LFS_NAME_MAX saves RAM.

LFS_FILE_MAX is useful as not all filesystem drivers support files > 2GiB since this will overflow signed integers (this driver _kinda_ supports files > 2GiB, it works but return values for lfs_file_seek are invalid).

But there's no cost associated with LFS_ATTR_MAX. Custom attributes don't have a RAM cost, and there's no runtime cost unless you use them, so changing LFS_ATTR_MAX doesn't impact the driver at all.

At least there's no value to LFS_ATTR_MAX that I've found yet, though it would be interesting if someone finds a use for it.

lsilvaalmeida, 2019-05-23:
Hi @geky, thanks for your response. 
I'm happy to help in this bug and the correction that you made seems good to me. 

>  I've considered dropping LFS_ATTR_MAX. It's only there for consistency with the other limits (LFS_FILE_MAX and LFS_NAME_MAX).

I think is nice to have these limits (it's classy haha) and after this correction, I don't see any trouble in maintain it. 

The issue is closed to me.

closed 2019-05-23 by lsilvaalmeida