littlefs-project/littlefs #191: LittleFS fails to open and save the data to the file

subhankhanm, 2019-05-28:
Hi,

I are trying to integrate LittleFS V2.0.4 on the iMXRT1052 microprocessor with QSPI Nor flash(IS25WP064A from ISSI) as filesystem storage on iMXRT EVKB development board (EVKB board modified), 
but somehow I can't get to make it work properly

The test I am performing is to open a particular file, read content(which is a uint32_t counter), increment the counter and write the counter value back to the file, then close the file. I am doing this every second. 

I am having multiple failures:
1) After mounting the LittleFS, it fails to create/open the file with the following error
(Failed to open file: 36). After 10-15 tries it then successfully creates/opens the file.
2) LittleFS fails to write(save) the counter value to the file. After a few successful reads and writes, littleFS fails to save the data to the file. As described earlier, I'm incrementing the counter value after every read and then saves the new value to the file. But on the next read, I am getting the same(previous) value.
```
Eg: 
count = 0
count = 1
count = 2
count = 3
count = 4
count = 5
count = 6
count = 7
count = 8
count = 9
count = 10
count = 11
count = 12
count = 13
count = 13
count = 13
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
count = 14
```

This is my config code:
```
const struct lfs_config cfg = {
        // block device operations
        .read  = lfs_read_wrapper,
        .prog  = lfs_write_wrapper,
        .erase = lfs_erase_wrapper,
        .sync  = lfs_sync_wrapper,

        // block device configuration
        .read_size = 256,
        .prog_size = 256,
        .block_size = 4096,
        .block_count = 2048,
        .lookahead = 256,
        .cache_size = 512,
       .block_cycles = 100,
};
```

lsilvaalmeida, 2019-05-28:
Hello @subhankhanm. Normaly the error code is a negative number. If you are casting it and the real error is -36, this means that your file name or path (root/directory_mame1/.../file_name) is too long. Maybe is there the problem. Can you show us more of your code ? 

Otherwise, are you shure that your wrapper functions are working well? You can try debug it to see if the problem is inside it.

subhankhanm, 2019-05-28:
Hi @lsilvaalmeida,

Thank you for your reply. 

The flash read wrapper function was taking much longer to read the contents from flash. Increasing the read spi frequency solved the issue.

closed 2019-05-28 by subhankhanm