littlefs-project/littlefs #195: LFS tries to program the same block twice without erase

kazink, 2019-05-29:
Hi, I tried to incorporate LFS to my project (STM32 + NAND Flash), but it failed on a simple test. From my debug output it looks like LFS tries to program the same block twice without erasing it in between.

My setup is:
read_size, prog_size = 2048,
block_size = 2048 * 64,
block_count = 1024,
cache_size, lookahead_size = read_size,
block_cycles = 1,
NO_MALLOC
static buffers defined accordingly to the documantation: read and prog buffer = 2048,
lookahead buffer = 2048, 64-bit aligned (checked).

1. Format the filesystem, then mount.
2. For x files:
- open file with RDWR | O_CREAT,
- write 1000 bytes,
- close file, open again with RDWR only, rewind,
- read data back and check verify it's the same as written,
- close file.

On first iteration (test_0) the file is written and read correctly, but on the second it fails.

My partial debug output (can provide full, if necessary);
------ test_1 ------
Opening file
NAND_device_read: block 585, off 0, page 37440, start 0, returned 0
NAND_device_read: block 587, off 0, page 37568, start 0, returned 0
NAND_device_read: block 587, off 2048, page 37569, start 0, returned 0
NAND_device_read: block 587, off 4096, page 37570, start 0, returned 0
NAND_device_prog: block 587, off 2048, page 37569, start 0, returned 0
NAND_device_read: block 587, off 2048, page 37569, start 0, returned 0
Writing file
Bytes written: 1000
Closing file
NAND_device_read: block 587, off 4096, page 37570, start 0, returned 0
NAND_device_prog: block 587, off 2048, page 37569, start 0, returned 0
NAND_device_read: block 587, off 2048, page 37569, start 0, returned 2
NAND_device_read: block 587, off 0, page 37568, start 0, returned 0
NAND_device_read: block 587, off 2048, page 37569, start 0, returned 2
### Close failed: -84 ###

Page and start are values I must pass to the physical NAND read/write procedures, so I have to calculate them from block and off that LFS supplies.
It's simply: 
page = block * 64 + off / 2048;
start = off % 2048;

Return value 0 means success, value 2 means unrecoverable ECC error.
No surprise, since LFS programmed the page 37569 on file open and on file close, without erasing the block in between. I also have NAND_device_erase output, so I would know.

lsilvaalmeida, 2019-05-29:
Hi @kazink. You are writing inline files (which ones are stored in the same block of theirs directory), so it's expected that you will write multiple files in the same block. But the page address should be different. So I was wondering if this offset is well calculated.

Otherwise, you have an ECC to check read errors? NAND Flash are really succetible to these errors and maybe is there the problem.

kazink, 2019-05-29:
Thanks for the answer.
Look at the "block" and "off" values - these are passed to the NAND function by LFS, and they are the same for both writes. So the calculations do not matter here (only "page" and "start" are calculated.

So you say, the two writes may be of the same data, but only ECC is wrong? I also tried ignoring ECC, but then on the second file LFS returned -2 (as if file did not exist).

Strangely enough, if I changed the file names (from "file_x" to "text_x"), LFS returned -2 only on 10-th file.

lsilvaalmeida, 2019-05-29:
Hmm, I had a similar problem when I wasn't using static buffers to the file cache. When you open the file, are you giving to it a static buffer (like in opencfg)?

kazink, 2019-05-29:
Yes, I do. The buffer is the same size as cache_size.

geky, 2019-05-29:
I need to look into this more, may be related to https://github.com/ARMmbed/littlefs/issues/188

> Return value 0 means success, value 2 means unrecoverable ECC error.

Does your block device callbacks return a positive 2? Unfortunately littlefs will break if the callbacks return a positive number. I need to change it to handle this better, but the block device should return a negative error code on error (LFS_ERR_CORRUPT would be perfect).

kazink, 2019-05-29:
No no, positive 2 is returned from my NAND driver. I then return LFS_ERR_CORRUPT in the translation layer read function.

kazink, 2019-06-01:
I have moved to a previous version 1.7, because it didn't have this problem, but it doesn't have bad block management and it's very slow sometimes. So I'm trying to give another shot at 2.0.

This is the data LFS writes two times to the same page, without erase:
```
NAND_device_prog:158: block 235, off 2048, size 2048
10 1f af f3 40 00 00 09 62 65 62 65 6c 65 5f 32
NAND_device_prog:158: block 235, off 2048, size 2048
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

It's not a surprise that ECC says there is something wrong.
Can you reproduce this?

What I'm doing is just writing 1000 files with consecutive names, 20 bytes each. 
The error appears at the file number 27.

kazink, 2019-06-01:
I have increased the write and cache sizes to twice the page size (2*2048), and it seems to work fine.

However the speed of it is horrible.
LFS 1.7 when writing 1000 files 20 bytes each, took max 350ms to open and close the file.
LFS 2.0 takes over 3 seconds to either open or close, and it's still 170th file (inline files disabled).

geky, 2019-07-01:
Hi @kazink, sorry it's taken me so long to get to your issue, I'm looking into it now. I can reproduce the 2048 B issue, unfortunately this hits the limit of an [internal field](https://github.com/ARMmbed/littlefs/blob/master/SPEC.md#0x0xx-lfs_type_name), so the problem is a bit entrenched.

Read/write sizes >1024 byte is a new area for littlefs, and unfortunately not tested well (sorry). No idea why 2*2048 works.

As for performance, I'm not sure why you're seeing worse performance for LFS 2.0, at worst case it should behave the save as LFS 1.7. (Though https://github.com/ARMmbed/littlefs/issues/158 may be the culprit).

Out of curiousity, how did you disable inline files? I thought I removed that option.

geky, 2019-07-01:
Just realized the terrible performance may be due to how prog_size=2048 and block_cycles=1 interact. If you increase block_cycles to something in the range of 100-1000 do you see better performance?

The block_cycles option doesn't exist in v1 because v1 doesn't have proactive wear-leveling. While block_cycles=1 provides the most strict uniform distribution of wear, it forces littlefs to relocate directory blocks every time they are full (instead of a cheaper garbage collection cycle).

Note that without O(n) RAM, this runtime hit is unavoidable in any filesystem.

To reduce the runtime impact, littlefs only relocates directory blocks every "block_cycles" garbage collection cycles. The improves the performance at a cost of no longer having perfect wear-leveling. However, the wear still approaches a uniform distribution, and the estimation of erase cycles on storage is already stochastic.

More info [here](https://github.com/ARMmbed/littlefs/blob/master/DESIGN.md#wear-leveling).

Normally this isn't so bad since these directory blocks are also performing their own form of static wear-leveling internally (logs wear-level naturally), and can be updated block_size/prog_size times before needing garbage collection. However, littlefs currently has an internal [log entry limit of 1024 bytes](https://github.com/ARMmbed/littlefs/blob/master/SPEC.md#0x5xx-lfs_type_crc). There is probably better actions to take if this limit is exceeded, but for now littlefs just defaults to a single log entry per directory block. The entire block is still utilized, but a garbage collection occurs on every file sync.

Possible solutions (wip):
1. Default to prog size for a max entry length field - The reason I'm not doing this now is because it creates a portability issue for drivers with different prog-sizes. Currently the prog-size has no impact on disk structures, and as such is not stored in the superblock info.
2. Increase log entry limit to 4096 - I did reserve 2-bits for this option, since I wasn't sure if those 2-bits would be more useful in the tag's size field (size of entries in dir) or id field (number of files per dir). It already seems that size is the best place to allocate these given the prevalence of 2048-4096 prog sizes on NAND.

geky, 2019-07-01:
Or, even better:

3. Pad commit with multiple CRC-tags until we reach the next prog_size boundary

This would actually solve all the issues with just a small code cost.

geky, 2019-07-01:
Ok, I've pushed up a fix for the workaround for the 2048 B prog_size issue: https://github.com/ARMmbed/littlefs/commit/7051bf58182e47dae115a3148b95825fc6c234a1

I want to at least add more testing around these large prog_sizes before merging that in, but let me know if it works for you.

kazink, 2019-07-09:
Thanks for the answer. Late is better than none :)
Unfortunately increasing the block_cycles even to 10k didn't help. The performance is slightly better than on 100, but not much.
The open and close times vary greatly (open 2-16 seconds, close 14ms-5sec), but they are too high for my purpose at just 250th file (I am testing by creating new files of random sizes from 1kiB to 1MiB).
I can post a test log with individual file times if you would like.

I think this comment is right on the reason for the slow down: https://github.com/ARMmbed/littlefs/issues/158#issuecomment-476578626
I have also noticed huge amounts of sectors read while creating a new file.

I have disabled inline files by commeting out the line 2740:
         if ((file->flags & LFS_F_INLINE) /* &&
            lfs_max(file->pos+nsize, file->ctz.size) >
            lfs_min(0x3fe, lfs_min(
                lfs->cfg->cache_size, lfs->cfg->block_size/8)) */ ) {
I thought it would disable iniling by forcing every inline file not to fit. .

The 2048 B prog_size solution works fine for disabled inline files.
But I have just uncommented the above block to check inlines too, and:
- with read/prog/cache sizes = 2048 the test fails at 1st file (lfs_close returns -84),
- with the sizes = 4096 the test fails at 3rd file (same result).

geky, 2019-07-09:
Oh, that is good to know. Thanks for posting an update and testing the block_cycles.

I'll have to see if I can reproduce the inline file issue. Inline files should at least help reduce the number of allocations.

Also the modification to remove inline files looks correct.

It may just be that littlefs isn't there yet for scalability. I'm surprised the issue is https://github.com/ARMmbed/littlefs/issues/158#issuecomment-476578626 as you only have 1024 blocks, but you're right the number of different sector reads strongly suggests it's related. Maybe the issue is exacerbated by the large read_size? Just thinking out loud, I don't have a better answer.