littlefs-project/littlefs #196: SPI Implementation for STM32L151 board. Do you have any drivers for SPI?

KRekha3, 2019-05-29:
i am using stm32l1xx_hal_spi.c in my project.  
I want to establish a communication between STM32L151 MCU and External flash memory using single SPI i.e 1 SI, 1 SO pin.
I could not find any example code. Please suggest.

geky, 2019-05-29:
The best resource on this would be the external flash's datasheet, it should document the opcodes you need to send over SPI for erasing/programming/reading:
http://www.macronix.com/Lists/Datasheet/Attachments/7478/MX25R2035F,%20Wide%20Range,%202Mb,%20v1.6.pdf

It may be useful to look at mbed's SPI driver, though the abstractions may make it a bit difficult to read:
https://github.com/ARMmbed/spif-driver/blob/78c4f98b808e86c50fe74f696ea8732d03e0b16b/SPIFBlockDevice.cpp#L274

I've also seen this guy before, may be useful, though I haven't had a chance to use it:
https://github.com/pellepl/spiflash_driver

KRekha3, 2019-05-29:
Thank you ..I will take a look

closed 2019-07-19 by KRekha3