littlefs-project/littlefs #197: Simplify attribute access on directory read?

e107steved, 2019-05-29:
I suspect I am not alone in using the new attribute capability to store file time and date.
This is already addressed for file-based accesses. However when enumerating a directory, the process for getting attributes is slightly long-winded, since the current file has to be looked up by name using lfs_getattr() - probably a bit slow with my I2C FRAM.
When calling lfs_dir_read(), would it be practicable to store the tag generated by lfs_dir_find() (or some other appropriate value) in the lfs_info structure, then have a function lfs_getattr_by_tag()? Looks as if it might be quite a small amount of additional code. Or is there a better way?

And on a slightly related note, how about defining some values for 'standard' attribute types? Various time and date types are obvious; possibly security-related ones as well.

geky, 2019-05-29:
Ah! Good to hear. And thanks for creating this issue, it's useful to know problems users run into.

Yeah, I think this is a part of a wider issue: in a traditional POSIX API, going from lfs_dir_read() -> other operations requires string manipulations. And string manipulations on a microcontroller is usually a sign of something going wrong.

Related: https://github.com/ARMmbed/littlefs/issues/42

I've been looking around at what else is out there, and one thing that's stood out is the openat APIs in Linux (similar to the `lfs_file_open_direct` suggested by @apmorton):
https://linux.die.net/man/2/openat

This would be a parallel set of APIs that take in a directory and a path instead of just a path. So attribute iteration could be turned into this:

``` c
lfs_dir_t dir;
err = lfs_dir_open(&lfs, &dir, "stuff");
if (err) {
    return err;
}

while (true) {
    struct lfs_info info;
    err = lfs_dir_read(&lfs, &dir, &info);
    if (err) {
        return err;
    }

    if (err == 0) {
        break;
    }

    uint32_t timestamp;
    err = lfs_getattrat(&lfs, &dir, info.name, TIMESTAMP_ATTR, timestamp, sizeof(timestamp));
    if (err < 0) {
        return err;
    }

    printf("file %s type %d time %d\n", info.name, info.type, timestamp);
}
```

That's what I'm currently leaning towards. Though as this duplicates the littlefs API in almost its entirety, I'm going to be cautious and see what others think.

geky, 2019-05-29:
I considered outlining some standard attributes, but the hard part is choosing formats that work for everyone. 40-bit timestamps for the 2038 problem? Is the Linux 8-bit permission field sufficient for cloud-connected devices?

I don't think I should be deciding these myself. I'd rather have some way we could propose and vote on standard attributes, but I don't have a good answer on how to do that at the moment.

e107steved, 2019-05-30:
Thankyou for the reply.  Your suggestion goes some way to simplifying things; however lfs_getattrat() would still have to search for the file entry; usually multiple reads, which are relatively expensive on a slow device.

My suggestion goes a bit further. First re-factor lfs_getattr() thus:
```

lfs_ssize_t lfs_getattr_by_tag(lfs_t *lfs, lfs_stag_t tag,
        uint8_t type, void *buffer, lfs_size_t size) {
    if (tag < 0) {
        return tag;
    }

    uint16_t id = lfs_tag_id(tag);
    if (id == 0x3ff) {
        // special case for root
        id = 0;
        int err = lfs_dir_fetch(lfs, &cwd, lfs->root);
        if (err) {
            return err;
        }
    }

    tag = lfs_dir_get(lfs, &cwd, LFS_MKTAG(0x7ff, 0x3ff, 0),
            LFS_MKTAG(LFS_TYPE_USERATTR + type,
                id, lfs_min(size, lfs->attr_max)),
            buffer);
    if (tag < 0) {
        if (tag == LFS_ERR_NOENT) {
            return LFS_ERR_NOATTR;
        }
        return tag;
    }

    return lfs_tag_size(tag);
}

lfs_ssize_t lfs_getattr(lfs_t *lfs, const char *path,
        uint8_t type, void *buffer, lfs_size_t size) {
    lfs_mdir_t cwd;
    lfs_stag_t tag = lfs_dir_find(lfs, &cwd, &path, NULL);
	
	return lfs_getattr_by_tag(lfs, tag, type, buffer, size);
}

```
The 'tag' value needed by lfs_getattr_by_tag() is presumably accessible, or easily computable, in lfs_dir_read(). So if this value is stored in the info structure, it becomes much more efficient to retrieve attributes. (This does assume I've understood enough about how things work under the hood!)

This could also minimise the amount of code added by a lfs_getattrat() routine.

On the attribute types, I suggest you approach it from a slightly different angle. Define a block of values which are reserved for standard attributes. Then define attribute type values for 'well-known' attributes. Unix and FAT time/date stamps come immediately to mind. None of this prevents users from defining their own types and formats; which they will hopefully put forward if they are of general use.