littlefs-project/littlefs #199: How to mount filesystem at particular address of flash memory

KRekha3, 2019-06-04:
Hi ,
i want to implement filesystem on external flash memory which is interfaced with STM32L151 controller.
flash is accessed through SPI lines. i have written below code for read and write operation.

`int user_provided_block_device_read(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, void *buffer, lfs_size_t size)
{
  printf("read\n");
  MEM_Read((block * c->block_size + off), (uint8_t*)buffer, size); //SPI read
  return 0;
}
int user_provided_block_device_prog(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, const void *buffer, lfs_size_t size)
{
//  c = (struct lfs_config *)0X700000;
  printf("prog\n");
  MEM_Write((block * c->block_size + off), (uint8_t*)buffer, size); //SPI write
  return 0;
}
const struct lfs_config cfg = {
    // block device operations
    .read  = &user_provided_block_device_read,
    .prog  = &user_provided_block_device_prog,
    .erase = &user_provided_block_device_erase,
    .sync  = &user_provided_block_device_sync,

    // block device configuration
    .read_size = 256,
    .prog_size = 256,
    .block_size = 32768,
    .block_count = 256,
    .cache_size = 256,
    .lookahead_size = 256,
    
    .read_buffer = lfs_read_buf,
    .prog_buffer = lfs_prog_buf,
    .lookahead_buffer = lfs_lookahead_buf   
};`

when i tried to debug filesystem takes const struct lfs_config *c = 0x00000000 as address. is there any way to change this address. because 0x00000000 is protected area which cannot be accessed by user.
Please suggest...

geky, 2019-06-10:
Hi @KRekha3, that shouldn't happen. I'm not sure what would cause that.

How are you calling `lfs_mount`?

KRekha3, 2019-06-10:
Thanks.
Issue is resolved ..problem was with the other code where we were trying to
integrate LFS. We identified correct location to insert the code. And also
with the configuration of filesystem. corrected the code to match the
requirement.
Now it is working.

KRekha3, 2019-06-10:
And also we erased flash memory completely and started with mount..it works
now.
thanks

geky, 2019-06-10:
Ah glad to hear!

closed 2019-06-19 by KRekha3