littlefs-project/littlefs #206: lfs error:969 Corrupted dir pair at 0 1

safirxps, 2019-06-14:


safirxps, 2019-06-14:
I got strange error at the begining of my software my code is as follow :

int filesystem_init(void)
{
  int err = 0;
  err = lfs_mount(&Filesystem_Handle, &Filesystem_Config);
  if(err != 0)
  {
   err = lfs_format(&Filesystem_Handle, &Filesystem_Config);
    if(err == 0)
      err = lfs_mount(&Filesystem_Handle, &Filesystem_Config);
  }
  return err;
}

using latest Littlefs version

my config is :

#define BASE_BLOCK_NUM 64 // this is the number of block on the lower side of the flash not included in Filesystem each Block is 4096 Bytes

#define READ_SIZE 1
#define PROG_SIZE 1
#define ERASE_SIZE 4096
#define APP_SIZE BASE_BLOCK_NUM*ERASE_SIZE
#define FLASH_SIZE_ 16384*ERASE_SIZE // There are 16384 blocks of 4096 Bytes in 64 MBytes Flash

const struct lfs_config Filesystem_Config =
{
  // block device driver context data
  .context = NULL,

  // block device operations
  .read = lfs_read_page,
  .prog = lfs_device_prog,
  .erase = lfs_erase_page,
  .sync = lfs_device_sync,

  // block device configuration
  .read_size = READ_SIZE,
  .prog_size = PROG_SIZE,
  .block_size = ERASE_SIZE,
  .cache_size = 256,
  .block_count = (FLASH_SIZE_-APP_SIZE) / ERASE_SIZE,
  .lookahead_size = 256,
  .block_cycles = 100,
};

lsilvaalmeida, 2019-06-14:
In the first mount is normal have this error because lfs can't found the superblock. But, after lfs_format, this error should go away. When are you having this error?

safirxps, 2019-06-14:
Hi,
Thanks it is working now.
New problem after filewrite, I did the following :

filesystem_filewrite("TestPol1",&polo,sizeof(testva));

with : 
int filesystem_filewrite(const char* filename, void *buffer, lfs_size_t size)
{
  lfs_file_t Filesystem_Filetype;
  int err = 0;
  err=lfs_file_open(&Filesystem_Handle, &Filesystem_Filetype, filename, LFS_O_WRONLY  | LFS_O_CREAT);
  if(err == 0)
  {
    lfs_size_t WriteSize=lfs_file_write(&Filesystem_Handle, &Filesystem_Filetype, buffer, size);
    err = WriteSize;
  }
  lfs_file_close(&Filesystem_Handle, &Filesystem_Filetype);
  return err;
}

The result of filewrite is ok, with the size of the block, however when calling file open to read the size just after it can not find the file, I look at in Flash with debugger , and I can find file name I was using, what can be wrong ?

lfs_file_open(&Filesystem_Handle, &Filesystem_Filetype, "TestPol1", LFS_O_RDONLY)  return file name not existing.

lsilvaalmeida, 2019-06-17:
Hello,
First of all, the size in lfs_file_write should be the size of data pointed by buffer. So, `filesystem_filewrite("TestPol1",&polo,sizeof(polo))` should be better. The same for reading. Maybe this wrong call of write has corrupted the file, I don't know.

asteriskSF, 2019-06-21:
> Hello,
> First of all, the size in lfs_file_write should be the size of data pointed by buffer. So, `filesystem_filewrite("TestPol1",&polo,sizeof(polo))` should be better. The same for reading. Maybe this wrong call of write has corrupted the file, I don't know.

This comment is not helpful.  The wrapper function from the reporter is correct usage of file write, the length is provided with the argument along with the pointer to buffer.  sizeof(buffer) would only return the size of a pointer, not the sizeof the data pointed to by the pointer.

KRekha3, 2019-07-02:
Hi @lsilvaalmeida , @safirxps 
even i am facinf similar issues, i have declared prog, read, lookahead size as 256bytes, if i try to write more than 256 it is corrupting mounted filesystem. what might be the mistake?

geky, 2019-07-09:
Hi @safirxps, thanks for creating an issue.

Are there any errors returned by `lfs_file_close`? The data usually not committed until `lfs_file_close`. You can commit earlier (and check for errors) with `lfs_file_sync` if you need to.

If there's no error, are you able to post a dump of the filesystem? I'm not sure what could be wrong.

Also what chip are you using? Does it support 4KiB erase blocks? I've seen a number of MCU internal flash require much larger erase blocks.

geniusgogo, 2019-07-18:
HI,I also encountered the same error.

> [I/SF] SPI Flash JEDEC-ID Detection(ID:c84017)
> [I/SF] SPI Flash Capacity:(8MB)
> lfs error:969: Corrupted dir pair at 0 1
> [I/INIT] File System format successful!
> [I/INIT] File System initialized!

> msh />echo "hello" hello.txt   
> msh />ls
> Directory /:
> hello.txt           5                        
> msh />cat he
> msh />cat hello.txt
> hellomsh />
> msh />

restart at this time.

> [I/SF] SPI Flash JEDEC-ID Detection(ID:c84017)
> [I/SF] SPI Flash Capacity:(8MB)
> lfs error:969: Corrupted dir pair at 0 1
> [I/INIT] File System format successful!
> [I/INIT] File System initialized!
> msh />ls
> Directory /:
> msh />

**Every time there is this problem with the restart**

geniusgogo, 2019-07-18:
but i switch littlefs v1.7 no error。

[I/SF] SPI Flash JEDEC-ID Detection(ID:c84017)
[I/SF] SPI Flash Capacity:(8MB)
[I/INIT] File System initialized!
msh />