littlefs-project/littlefs #212: Typos in lfs.h

joel-felcana, 2019-06-20:
I think I've found some inconsistencies in the documentation in lfs.h:

line 545 lfs_file_rewind
> // Change the position of the file to the beginning of the file
//
// Equivalent to lfs_file_seek(lfs, file, 0, LFS_SEEK_CUR)
// Returns a negative error code on failure.
int lfs_file_rewind(lfs_t *lfs, lfs_file_t *file);

I think it's meant to say "Equivalent to lfs_file_seek(lfs, file, 0, **LFS_SEEK_SET**)"

line 527, lfs_file_seek
> // Change the position of the file
//
// The change in position is determined by the offset and whence flag.
// Returns the old position of the file, or a negative error code on failure.

lfs_file_seek returns the new position of the file.

Johnxjj, 2019-06-24:
I very much agree with you! In addition, i find some other problems.

	lfs_file_opencfg(&lfs, &file[0], "boot1", LFS_O_RDWR | LFS_O_CREAT, &file_cfg);
	lfs_soff_t position = lfs_file_tell(&lfs, &file[0]);
	LFS_DEBUG("lfs_file_tell. ( lfs_file_tell: %d )", position);
	lfs_soff_t file_size = lfs_file_size(&lfs, &file[0]);
	LFS_DEBUG("lfs_file_tell. ( lfs_file_size: %d )", file_size);	
	position = lfs_file_seek(&lfs, &file[0], 10, LFS_SEEK_SET);
	position = lfs_file_tell(&lfs, &file[0]);
	LFS_DEBUG("lfs_file_tell. ( position: %d )", position);
	position = lfs_file_seek(&lfs, &file[0], 5, LFS_SEEK_CUR);
	position = lfs_file_tell(&lfs, &file[0]);
	LFS_DEBUG("lfs_file_tell. ( position: %d )", position);

	position = lfs_file_seek(&lfs, &file[0], -3, LFS_SEEK_END);
	position = lfs_file_tell(&lfs, &file[0]);
	LFS_DEBUG("lfs_file_tell. ( position: %d )", position);

Run these codes, the log will be:
       lfs debug:504: lfs_file_tell. ( lfs_file_tell: 0 )
       lfs debug:507: lfs_file_tell. ( lfs_file_size: 5 )
       lfs debug:511: lfs_file_tell. ( position: 10 )
       lfs debug:515: lfs_file_tell. ( position: 15 )
       lfs debug:519: lfs_file_tell. ( position: 2 )

LFS_SEEK_END means "Seek relative to the end of the file ", So i think  after code "position = lfs_file_seek(&lfs, &file[0], -3, LFS_SEEK_END);". Shouldn’t this position be:
 position =  position + size -3 =15+5-3=17.

Or I have a wrong understanding.

joel-felcana, 2019-06-24:
Not entirely sure about the output that you get. 

When you first open your file in RW mode with the create flag, you get the expected 0 from the first `lfs_file_tell()`, but then you call `lfs_file_size()` and you say that you expect a 5 from it, but I can't see how, given that you haven't read or write from the file, so it should output 0 too. From there, yes, the first seek puts you in 10, as LFS_SEEK_SET is absolute file position, then add 5 to that from your seek with LFS_SEEK_CUR and get a 15. 

About the last one, I get 15, not 2. If you check the code, you will notice that when using the LFS_SEEK_END, it returns a position applying the offset to the value of the `ctz.size` property of the file. That property is not touched by seek or tell, so I would say that what you are doing is not supported (that is, moving past the limits of a file and expecting the seek operations to work).

This is the output I get from your code, for your reference
```c
I (   278) │ lfs_file_tell. ( lfs_file_tell: 0 )
I (   283) │ lfs_file_tell. ( lfs_file_size: 0 )
I (   287) │ lfs_file_tell. ( position: 10 )
I (   292) │ lfs_file_tell. ( position: 15 )
I (   296) │ lfs_file_tell. ( position: 15 )
```

Johnxjj, 2019-06-25:
Hi @joel-felcana , i am very sorry that I missed a point: Before these codes, i have created a file and wrote 5 bytes of data. So lfs_file_size will be 5.

Johnxjj, 2019-06-25:
I already know where I am wrong, please forgive me for the first time I am in contact with the file system. I understand the comment "Change the position of the file" in lfs_file_seek as the location of the file instead of the position of the pointer *file.

joel-felcana, 2019-06-25:
Well, about the 5 of your second output, then it's settled.

>  I understand the comment "Change the position of the file" in lfs_file_seek as the location of the file instead of the position of the pointer *file.

I think that's how it's meant to be understood. If you move to position 5, reading or writing 1B will return / write the 6th byte (0-index) of the file. As long as you don't get out of bounds, everything should work as expected. If you just open a new empty file and move to position 50, you will definitely break things.

geky, 2019-07-03:
Good finds, thanks for creating an issue! Created a pr https://github.com/ARMmbed/littlefs/pull/221

I'm copying the Open Group/POSIX standard where possible, which do have some quirks, but means I have to worry less about missing functionality:
https://pubs.opengroup.org/onlinepubs/009695399/functions/lseek.html

> ```
> lfs debug:519: lfs_file_tell. ( position: 2 )
> ```
>
> LFS_SEEK_END means "Seek relative to the end of the file ", So i think after code "position = lfs_file_seek(&lfs, &file[0], -3, LFS_SEEK_END);". Shouldn’t this position be:
>
> ```
> position =  position + size -3 =15+5-3=17.
> ```

Position = 2 is correct. The key is that file position does not change the end of the file until there is a write.

```
position = size - 3 = 5 - 3 = 2
```

> That property is not touched by seek or tell, so I would say that what you are doing is not supported (that is, moving past the limits of a file and expecting the seek operations to work).

Surprisingly, it is. Seeking allows the position to go beyond the end of the file.

In more complex filesystems this can be used to create "holes" in files. littlefs doesn't support holes, so if you write past the end of a file it will just fill the area between with zeros.

This can create some confusing situations though. Notably the file "end" never changes unless you call lfs_file_write.

---

At least that's how it's _supposed_ to work. If you find differences let us know as it could be a bug.