littlefs-project/littlefs #215: Strange lfs_file_write() return value

ksstms, 2019-06-26:
Hi!

I'm using LFS v2.0.0 for storing some 16MB firmware files on an SPI NAND flash. I experienced no problems on known good flash devices, but now I'm testing it on a chip with a few bad blocks.
I have assertions and all kinds of logs enabled. Sometimes I see block relocations in the log, but the file system works fine. But sometimes writes are failing for no apparent reason.

For testing I have tried to write the files in 32 byte operations _(yes, terribly small buffer, I'm working on it)_. 
At about 1 MB `lfs_file_write()` returns 50 for a 32 byte write. The log is empty. If I ignore this and keep writing and read the file back, there are some missing parts in the file and the size is about 400 bytes smaller than it should be.

Do you have any suggestions on debugging this issue?

**Flash chip info:**
- Page size: 2176 bytes (2048 + 128 bytes)
- Block size: 64 pages (128K + 8K bytes)
- Plane size: 1Gb (1 plane, 1024 blocks per plane)
- Erasable per block. Should be written by page for the on-chip ECC to work

**LFS config**
 * Read size: 1              
 * Program size: 2048
 * Block size: 64*2048
 * Block count: 1024   
 * Block cycles: 10  
 * Cache size: 2048     
 * Lookahead size: 64
 * Max file size: 20\*1024\*1024 (20 MB)
 * The others are default, and the buffers are allocated manually

ksstms, 2019-06-27:
My SPI flash driver works fine indeed, but the erase wrapper function did not convert its error code correctly, so LFS did not receive LFS_ERR_CORRUPT. I'm sorry for bothering you.

closed 2019-06-27 by ksstms

geky, 2019-07-01:
Out of curiosity, was your SPI flash driver returning positive error codes? Currently LittleFS breaks if that happens, as it only expects negative error codes (related https://github.com/ARMmbed/littlefs/issues/169).

ksstms, 2019-07-02:
Yes, it does. 
The problem was that I forgot to convert it to LittleFS compatible codes in my erase function.