littlefs-project/littlefs #216: LittleFS usage question

felipeLeast, 2019-06-26:
Hello, 

I'm starting a implementation of LittleFS onto my code and in the process some questions were raised:

1- Is LittleFS thread safe? I mean, can I mount my fs in my main and use within any threads with no worries, or should I implement a mutex to work with fs?
2- Should I unmount my filesystem or can I live it mount through out the entire program?
3- Do I have a limit of files/folders I can create? If not, does it mean that by increasing the number of file/folders will increase the RAM usage? Can I see a metrics of it?
4- In my application I have 10 files that won't change much, but I have to consistently write 8 bytes do flash, every 5 minutes approximately. What would be the best implementation, thinking on wear and RAM usage, use multiple files for each write sepparating them on folders, or use a single file and just append to the end the next data?

Thank you in advance,

geky, 2019-07-02:
Hi @felipeLeast, hopefully these answers help:

1. If you're accessing LittleFS from different threads, you will need to protect LittleFS with a mutex. But all you need is a mutex and then LittleFS will be completely thread safe, even if you open multiple files and write to them in different threads.

2. The only reason to unmount is if you want to free up RAM for another task. Because it's power resilient you may never need to call unmount at all!

3. No limits other than the available storage. LittleFS was designed intentionally to separate concerns for RAM and external storage. The only RAM cost is if you have multiple files open simultaneously.

   There is a runtime cost where the block allocator runs linearly with the number of files. This is something I'm hopeful we can improve, but that how it works right now.

4. This sounds like you are storing a log of sorts? The only thing to watch out for is that LittleFS currently has really bad performance with random writes. So this is currently the best way to implement a LittleFS-friendly log:
https://github.com/ARMmbed/littlefs/issues/104#issuecomment-423605528

   Though I haven't compared it to storing single files per update, which may take better advantage of the recently added inline files. It may be worth trying it to find out, but I haven't had a chance myself.

felipeLeast, 2019-07-03:
Hi @geky,

Yes it does! Thank you!

About the LogRotate I actually was implementing something very similar before your answer, only difference is that I don't have a limit on how many log files I have, and every time I send a file up it deletes all the old files.

The only follow up question is, I'm currently using fopen for all my files interactions, is there any difference from that to lfs_file_open?

Thank you,

geky, 2019-07-03:
LittleFS doesn't provide an fopen by itself, so I'm assuming this is Mbed OS? fopen is provided by the toolchain's retargeting layer, while lfs_file_open is the direct call.

With fopen there are more layers of indirection that each pull in their own code costs, but a benefit is you can change the underlying filesystem easier (to FAT for example).

closed 2019-08-09 by felipeLeast