littlefs-project/littlefs #217: Some of my littlefs performance numbers

kazola, 2019-06-26:
Dear all,

I hope these tests I did yesterday on my ARM MCU and littlefs are useful to give somebody approximate time values for some basic operations with littlefs. I have a 8192 bytes FRAM SPI memory connected.

Consider the following numbers as:
(read_size, program_size, block_size, block_count, cache_size, lookahead_size, spi_freq_MHz)

Examples come next:
No littlefs, time to low-level write 256 bytes via SPI:
(_, 1) = 2 ms
(_, 6) = 0.35 ms
(_, 12) = 0.17 ms

Littlefs format function:
(16, 16, 256, 32, 16, 16, 1)    = 12.7 ms
(16, 16, 256, 32, 16, 16, 6)    = 8.9 ms
(16, 16, 256, 32, 16, 16, 12)  = 6 ms

(64, 64, 256, 32, 64, 16, 1)    = 10 ms
(64, 64, 256, 32, 64, 16, 6)    = 5 ms
(64, 64, 256, 32, 64, 16, 12)  = x

(256, 256, 256, 32, 256, 16, 1) = 15.6 ms
(256, 256, 256, 32, 256, 16, 6) = 5.32 ms
(256, 256, 256, 32, 256, 16, 12) = x

= x means console output a "Superblock 1 become unwritable" error.

Littlefs boot_count example:
(16, 16, 256, 32, 16, 16, 1)    = 14 ms
(16, 16, 256, 32, 16, 16, 6)    = 15 ms

(64, 64, 256, 32, 64, 16, 1)    = 17.5 ms
(64, 64, 256, 32, 64, 16, 6)    = 24.5 ms

(256, 256, 256, 32, 256, 16, 1) = 21.3 ms
(256, 256, 256, 32, 256, 16, 6) = 9.2 ms

Littlefs create a 256 bytes file:
(16, 16, 256, 32, 16, 16, 1)    = 55 ms
(16, 16, 256, 32, 16, 16, 6)    = 50 ms

(64, 64, 256, 32, 64, 16, 1)    = 53.5 ms
(64, 64, 256, 32, 64, 16, 6)    = 23.1 ms

(256, 256, 256, 32, 256, 16, 1) = 36 ms
(256, 256, 256, 32, 256, 16, 6) = 16 ms

Please let me know if these are useful to you. Also, please let me know if some configuration is not worth mentioning so we can remove it. Finally, if someone has some explanation for the error given at 12 MHz, it will be very welcome.

Have a nice day.

asteriskSF, 2019-06-27:
It would be interesting to see these numbers in terms of number of erase operations and write or read operations.   IIRC, FRAM  is extremely fast and relatively new technology.  Other technologies are bottlenecked by erase operations and write operations.

geky, 2019-07-03:
> Finally, if someone has some explanation for the error given at 12 MHz, it will be very welcome.

Is it possible the 12 MHz is too fast for the FRAM part somehow? The combination of 256 B + 12 MHz results in the largest bandwidth of program data going over the bus. This could be resulting in lost bytes.

kazola, 2019-07-07:
Sorry, this week I could not work on this.
MB85RS64V maximum frequency is 20 MHz so it should be able to do it.
Yes, I plan to do the same numbers for a NAND memory when I receive it. I will include some erase operations.

kazola, 2019-08-08:
Maybe  6 MHz is a limitation of my hardware, was wiring this through a prototype breadboard. Will try with real PCB boards soon.

closed 2021-11-30 by kazola