littlefs-project/littlefs #218: a strange problem when the size of file is less than or equal to cache_size

Johnxjj, 2019-06-27:
I found a strange problem. When the size of file size is less than or equal to cache_size and does not exceed 512 bytes, this file will be stored in the 0, 1 super block; otherwise, it will be stored in the data block. So when I try to modify this file, for example, assign a new value to the 512 bytes of the file, these new values ​​are still stored in the super block, then I am going to add 512 bytes to the end of the file. Make it length to 1024 bytes, then littlefs will reopen a new data block (non-super block), and copy the re-assigned 512 bytes in the super block to the new data block, plus the last end 512 bytes. These operations have no effect on the correctness of the contents of the file.
      However, imagine a situation and continue to write new files. The super block will continue to record the file log after the 512 bytes written earlier, and when the super block is full, the contents of the super block 0 will be copied to the super block 1. I observed from the log: The copied data includes the junk file in super block 0, and in the subsequent copy (copy to super block 0), the
[littlefs.txt](https://github.com/ARMmbed/littlefs/files/3334882/littlefs.txt)
 junk file always exists, so this is a bit strange. 

	int times_trade = 0;
	while (1)		//
	{
		times_trade++;
[超级块记录.txt](https://github.com/ARMmbed/littlefs/files/3334887/default.txt)

		LFS_DEBUG("times_trade. ( times_trade: %d ))\n", times_trade);

		//Trading_flow
[littlefs.txt](https://github.com/ARMmbed/littlefs/files/3334883/littlefs.txt)

		LFS_DEBUG("\nStatic_file. LINE: %d\n", __LINE__);	
		memset(&file_write_buf, ((file_data++)%256), sizeof(file_write_buf));
		static int Trading_cycle_count = 0;
		EA_ucLFSWriteFile(&lfs, &file, "Trading_flow", file_write_buf, 512, Trading_cycle_count *512, LFS_SEEK_SET);
		memset(&file_write_buf, ((file_data++) % 256), sizeof(file_write_buf));
		EA_ucLFSWriteFile(&lfs, &file, "Trading_flow", file_write_buf, 512, Trading_cycle_count *512, LFS_SEEK_SET);
		Trading_cycle_count++;
		if (Trading_cycle_count == 300)
		{
			Trading_cycle_count = 0;
		}	
		LFS_DEBUG("Trading_flow. ( size: %d KB ))\n", lfs_file_size(&lfs, &file) / 1024);
       }

Johnxjj, 2019-06-27:
`int times_trade = 0;
while (1)		//
{
	times_trade++;
	LFS_DEBUG("times_trade. ( times_trade: %d ))\n", times_trade);
	//Trading_flow
	LFS_DEBUG("\nStatic_file. LINE: %d\n", __LINE__);	
	memset(&file_write_buf, ((file_data++)%256), sizeof(file_write_buf));
	static int Trading_cycle_count = 0;
	EA_ucLFSWriteFile(&lfs, &file, "Trading_flow", file_write_buf, 512, Trading_cycle_count *512, LFS_SEEK_SET);
	memset(&file_write_buf, ((file_data++) % 256), sizeof(file_write_buf));
	EA_ucLFSWriteFile(&lfs, &file, "Trading_flow", file_write_buf, 512, Trading_cycle_count *512, LFS_SEEK_SET);
	Trading_cycle_count++;
	if (Trading_cycle_count == 300)
	{
		Trading_cycle_count = 0;
	}	
	LFS_DEBUG("Trading_flow. ( size: %d KB ))\n", lfs_file_size(&lfs, &file) / 1024);
 }
`

geky, 2019-07-10:
Good find! You're right this shouldn't happen and it's a bug. littlefs is supposed to clean up those inline files during compaction, but that logic was broken. It went undetected since the fetch logic will discard the duplicate when it sees a more recent tag.

I'm currently looking into the best way to fix this, littlefs can either 1. discard inline files during compaction, or 2. explicitly tag inline files for deletion.

1 is more efficient, but 2 leaves open the possibility to add support for files that are stored in both inline and skip-list parts. But I'm not sure the idea is possible even with 2 and it adds complexity.

1. here: https://github.com/ARMmbed/littlefs/tree/fix-file-uniqueness1
2. here: https://github.com/ARMmbed/littlefs/tree/fix-file-uniqueness2

geky, 2019-07-18:
Going with option 1. I realized there was already a plan for inline+ctz files in place. If we want both we can repurpose the device-side-only tag LFS_TYPE_FROM to actually give us 256 possible inline "chunks" for both inline and ctz files.

closed 2019-07-19 by Johnxjj