littlefs-project/littlefs #219: find the existing files

friguimahdi, 2019-07-01:
Hello, 

Is there any possible way to know what are the files that exists ? (name at least)

friguimahdi, 2019-07-02:
Its done using lfs_dir_read

closed 2019-07-02 by friguimahdi

Johnxjj, 2019-08-21:
if we don't know the file name,  how to know what are the files that exists?

friguimahdi, 2019-08-21:
you need to open the target directory that contains the files then you just read it using a while loop.

ret = lfs_dir_open(&lfs,&dir,"/");
   do
   {
    lfs_dir_read(&lfs,&dir,&info);
    if(info.name[0]!='.')
    {
    BSP_LCD_DisplayStringAtLine(i,(uint8_t*)info.name); /* Displaying the file names at line i */
    i++;
    }
   }while(info.name[0]!='\0');
   ret = lfs_dir_close(&lfs,&dir);

Johnxjj, 2019-08-22:
Thank you for the code provided very much, which helped me a lot!