littlefs-project/littlefs #220: Problem with lfs_file_seek

klausbng, 2019-07-02:
Hello,
if I use the following code:
    lfs_t g_sLfs;
    tsFile g_sFileObject;
    char buffer[16];
    lfs_soff_t offs;
    int ret;
    ...
    lfs_file_open(&sLfs, &sFileObject, "filename", LFS_O_RDONLY);

    offs = lfs_file_seek(&sLfs, &sFileObject, 0x100, LFS_SEEK_SET);
    ret = lfs_file_read(&sLfs, &sFileObject, buffer, 16);

    offs = lfs_file_seek(&sLfs, &sFileObject, 0x2000, LFS_SEEK_SET);
    ret = lfs_file_read(&sLfs, &sFileObject, buffer, 16);

the second lfs_file_seek is ignored and lfs_file_read reads at the position just after the previous read. However, lfs_file_seek returns the correct value.

If I look into the code of lfs_file_read, I can see, that only on first read LFS_F_READING is checked. In this case file->off will be set by lfs_ctz_find to the position inside the current block. I don't know the difference between file->off and file->pos, but I assume, that file->off is the position offset within the current block. If I now call a second lfs_file_seek, LFS_F_READING is already set, so the next read reads within the same block at file->off although the new value of file->pos should lead to a new load of a block. 
A workaround could be to clear the LFS_F_READING-flag within lfs_file_seek. I don't know, if this fixes all issues with a change of the file->pos value, maybe also other function modify it? And if I seek to position within the same block, it would be unnecessarily re-read. This would be a small performance impact, but I think we could live with it.

Regards
Klaus

lsilvaalmeida, 2019-07-05:
Hello @klausbng. I think file->pos is equal to file->off if its size is smaller than block_size (in my mind, file->pos is an absolute file offset and file->off is an local offset, like you said). 
I had a similar issue using file seek, but my mistake was to try to read in a position "outside" my file (pos >= file->ctz.size). So, are you sure that your file is gratter than 0x2000? Maybe there is a bug in the fils->ctz.size, idk.

klausbng, 2019-07-06:
Hello @lsilvaalmeida. The file has a size of about 60kByte, so there are several pages used. I try to seek on a position within the file and the read-size is also small enough. I have debugged the code - it is impossible that it works if I am doing a seek to a position into a new page. I have tried my workaround to clear the LFS_F_READING flag. Up to now everythings works, I have not found any drawback. Under some circumstances, however, it might have some performance impacts, if I always seek into the same page. I am sure, this could be done in a better way, but I didn't have the time to implement it.

Regards
Klaus

geky, 2019-07-09:
Hi @klausbng, thanks for creating an issue. Are you able to also reproduce this one with emubd? I tried but can't seem to reproduce this behavior. Though if it's an issue with LFS_F_READING it's very likely that it's a result of operations on the file before it got to this point.
https://gist.github.com/geky/28ff9f807fe7f8060acb8f02b42670a1

Also thanks @lsilvaalmeida, you're both right that off = offset in block, and pos = position in file.

File positions greater than the file size are legal, `lfs_file_read` should return 0 in that case, similar to `lfs_file_read` at the end of a file.

`lfs_file_seek` always calls `lfs_file_flush` ([here](https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L2825)), which is responsible for clearing the LFS_F_READING flag and clearing the cache LFS_F_READING protects ([here](https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L2489-L2494)). Though this is an only an issue if you later use the file to write.

geky, 2019-07-09:
Actually, which version of littlefs are you on? (`git log --oneline -n1`)

This may have actually been fixed in v2.0.5: https://github.com/ARMmbed/littlefs/issues/172

klausbng, 2019-07-10:
Yes, it seems it is fixed in 2.0.5. I tested that version and I even saw in the debugger that now LFS_F_READING is cleared in lfs_file_seek.

closed 2019-07-10 by klausbng

geky, 2019-07-12:
Good to hear 👍

muratdemirtas, 2022-07-17:
its seems like not fixed, littlefs is hangs at           

 while (file->pos < file->ctz.size)