littlefs-project/littlefs #222: LFS corrupted after 256 bytes of write

KRekha3, 2019-07-04:
Hi,
i have below configurations:
`uint8_t lfs_read_buf[256];
uint8_t lfs_prog_buf[256];
uint8_t __attribute__((__aligned__(8))) lfs_lookahead_buf[256] ;
uint8_t lfs_file_buf[256];

// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
  // block device operations
  .read  = &user_provided_block_device_read,
  .prog  = &user_provided_block_device_prog,
  .erase = &user_provided_block_device_erase,
  .sync  = &user_provided_block_device_sync,
  // block device configuration for 8Mbytes memory
  .read_size = 256,
  .prog_size = 256,
  .block_size = 32768,
  .block_count = 256,
  .cache_size = 256,
  .lookahead_size = 256,
  .attr_max = 1022,
  .file_max = 8388608,
  .block_cycles = 100,
  .read_buffer = lfs_read_buf,
  .prog_buffer = lfs_prog_buf,
  .lookahead_buffer = lfs_lookahead_buf
};`
i am trying to write to file using lfs_file_write(), after 256 bytes written to file , LFS tries to relocate file and erases the flash address where previous tags and metadata was written. below is screenshot of code:
[image]

How to resolve this? any changes to be made with respect to configuration?

lsilvaalmeida, 2019-07-05:
It is normal. If your file is grater than your cache (or 1/4 of block size), lfs will find a new free block to write to and this file will not be anymore an inline one. 
If lfs can't find a free block, there is an issue. Some questions that can help you to find your problem: 
- you really need a 256B lookahead buffer (it means, do you have more than 8*256 blocks in your device)? Maybe lfs is just trying to fill this buffer, but as your lookahead is so large, the fs take too much time to do it. 
- are you sure that lfs is trying to erase a metadata block?

KRekha3, 2019-07-08:
Thanks for response @lsilvaalmeida, 
just to test i have changed lookahead buffer size to 64 (I have 64*8 blocks in device)and i am still facing the same issue.
LFS corrupts after writing some bytes (it is not consistent).
"are you sure that lfs is trying to erase a metadata block?" - i mean if we mount lfs @0x00000000, the same address is being erased after writing few bytes to flash memory.

lsilvaalmeida, 2019-07-08:
> I have 64*8 blocks in device
The config shows that you have 256 blocks (32*8).

> LFS corrupts after writing some bytes
It returns with LFS_ERR_CORRUPT or another error code?

Are you passing `lfs_file_buf` to the fs (i.e. are you using `lfs_file_opencfg`)? Or are you using a malloc to allocate this cache? 

Just for curiosity, yours underlying functions (.read, .prog, etc) return 0 after a successful read/write/etc and something different than zero if not?

KRekha3, 2019-07-08:
I have changed lookahead buff size to 64(64*8 blocks as per your previous
comment).
I am using malloc to allocate it.
I will come back to you on the error code type.

KRekha3, 2019-07-09:
@lsilvaalmeida  , after few bytes of write operation, if i try to mount lfs it throws -84 as error code.
[image]

attached screenshot of the same. 
below is the configuration:
[image]

geky, 2019-07-09:
Hi @KRekha3, thanks for raising an issue, also thanks @lsilvaalmeida.

From your other issues, I believe you are using the MX25R? The MX25R has a minimum erase block of 4096 bytes, so your block_size must be a multiple of 4096. That might be the issue.

Also you mentioned you are using dynamic memory, if you set read_buffer/prog_buffer/lookahead_buffer to NULL, littlefs will dynamically allocate the correct amount. This may be a good way to sanity check that the buffers are set up correctly.

KRekha3, 2019-07-10:
Thanks I will check and get back on this.

KRekha3, 2019-07-10:
i have changed configuration with block size = 4096, i tested this before also also. but after making read, prog, file buffer as null and changed block size to 4kb , but stil LFS is corrupted.

[image]

`char test_arr[4096];
            //static int Fileerr2 = 0;
            string_t fName[] = "test1.bin";

            for (int i =0; i< sizeof(test_arr);i++){
                test_arr[i] = i;
            }

            if (lfs_mount(&lfs, &cfg)) {
              ASSERT(FALSE);
              // reformat if we can't mount the filesystem
              // this should only happen on the first boot
              if ( lfs_format(&lfs, &cfg) ) {
                  printf("File not formatted\n");
                  ASSERT(FALSE);
              }
              if (lfs_mount(&lfs, &cfg)) {
                  ASSERT(FALSE);
              }
            }

            /* create and write/read (boot_count) file */
            printf("File open\n");
            if(lfs_file_open(&lfs, &file, fName, LFS_O_RDWR |LFS_O_APPEND | LFS_O_CREAT)) {
                ASSERT(FALSE);
            }
            int32_t fsize = lfs_file_size(&lfs, &file);
            printf("fsize = %d\n", fsize);
            
            memset(test_arr,'z',sizeof(test_arr));
            if (sizeof(test_arr) != lfs_file_read(&lfs, &file, &test_arr, sizeof(test_arr))) {
                ASSERT(FALSE);
            }
            
            printf("File Write\n");
            if (sizeof(test_arr) != lfs_file_write(&lfs, &file, test_arr,4096)) {
                ASSERT(FALSE);
            }
            
            fsize = lfs_file_size(&lfs, &file);
            printf("fsize after write = %d\n", fsize);
            printf("File Close\n");
            if (lfs_file_close(&lfs, &file)) {
                ASSERT(FALSE);
            }
     
            if ( lfs_unmount(&lfs) ) {
                ASSERT(FALSE);
            }`

1st time - LFS writes 4096 bytes successfully 
2nd time- LFS mount works properly and starts writing 4096 bytes more to file.
3rd time - LFS mount corrupted, LFS tries to format the filesystem

KRekha3, 2019-07-18:
I have resolved the issue. there was a mismatch in the block_size and erase function used. erase function was erasing more than expected area due to which i was getting error.

geky, 2019-07-18:
Ah! Glad to hear. That sounds like it was a nasty bug to track down.

closed 2019-07-18 by geky

KRekha3, 2019-07-18:
Yes that's true

geky, 2019-07-18:
I went ahead and closed the issue, but feel free to reopen or create another issue if you run into any more problems. Best of luck 👍

KRekha3, 2019-07-18:
Sure. Thank you very much