littlefs-project/littlefs #223: Strange emulation test failure with actual device config

ucayalifish, 2019-07-08:
Hi,
First, thanks for your work, it is really great!
We encounter an interesting emudb tests failure. The context is as following.
We are trying to use littlefs in our project for saving a number of data arrays just for persistence at device shutdown and reading them at device restart. Some of the files are quite large i.e. up to 15MB but overall amount of data is within 24-32 MB.
We use 512 MB raw serial nand (2048 bytes page, 512 bytes minimum write ECC size, 64 page per block, 4096 blocks altogether) with the following config

prog_size: 512 bytes
read_size: 2048
block_size: 64*page_size
block_count: 4096
cache_size: 2048
lookahead buffer: 4096/8
block_cycles: 100

As we've encountered some problems with littlefs, we run the emubd tests with our settings. The "Chained dir exhaustion test" failed consistently with message:
`lfs: lfs.c:85: lfs_bd_read: Assertion `block < lfs->cfg->block_count' failed.
Traceback (most recent call last):
  File "tests/test.py", line 61, in <module>
    main(*sys.argv[1:])
  File "tests/test.py", line 58, in main
    execute()
  File "tests/test.py", line 44, in execute
    subprocess.check_call(["./lfs"])
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['./lfs']' returned non-zero exit status -6
Makefile:51: recipe for target 'test_alloc' failed
make: *** [test_alloc] Error 1
`
Additional debug output reveals that block var happens to be 1634235497. The test completes OK if we change block_count to 1024, 2048 or 8192, the problem takes place only if the block number is 4096.
Unfortunately, right now we do not have time to investigate the issue any deeper thus just report it.

Thank you for advance.

lsilvaalmeida, 2019-07-08:
Hello!
I think prog_size should be a multiple of read_size, but I don't know if this is the source of your problem. It seems that lfs fails in allocate a new block. BTW, lfs was not made for a so large device, then it can present some scalability issues.

geky, 2019-07-08:
Hi @ucayalifish, thanks for creating an issue

Hmmm, could this be related to https://github.com/ARMmbed/littlefs/issues/195?

I've pushed up a workaround here: https://github.com/ARMmbed/littlefs/commit/7051bf58182e47dae115a3148b95825fc6c234a1

Currently working on a better fix/tests, but let us know if that helps

ucayalifish, 2019-07-08:
Hi,
Thanks for your response!
I'm not sure this is connected to #195 so I'll try to apply the workaround and check. It definitely looks like either a kind of overflow or misalignment but I cannot think it out right now.

geky, 2019-07-08:
It also could be an issue with prog_size being smaller than read_size as @lsilvaalmeida mentioned.

It's _technically_ not a requirement, but it's very poorly tested because of how uncommon this configuration is (we need coverage info).

Out of curiousity what device is this that can't read in the same units that it can write in? As a workaround you could try setting the prog_size = read_size. This would decrease the number of writes per block but would at least narrow down if this is indeed the issue.

ucayalifish, 2019-07-11:
Hi,
Sorry for delay

We've tested branches master, fix-lfs-seek-typos, fix-file-uniqueness1 and fix-file-uniqueness2 with the following configuration:

`-uint8_t buffer[1024];
-uint8_t wbuffer[1024];
-uint8_t rbuffer[1024];
+uint8_t buffer[1024 * 2];
+uint8_t wbuffer[1024 * 2];
+uint8_t rbuffer[1024 * 2];
 lfs_size_t size;
 lfs_size_t wsize;
 lfs_size_t rsize;
@@ -62,31 +62,31 @@ lfs_size_t rsize;
 uintmax_t test;

 #ifndef LFS_READ_SIZE
-#define LFS_READ_SIZE 16
+#define LFS_READ_SIZE 2048
 #endif

 #ifndef LFS_PROG_SIZE
-#define LFS_PROG_SIZE LFS_READ_SIZE
+#define LFS_PROG_SIZE 512
 #endif

 #ifndef LFS_BLOCK_SIZE
-#define LFS_BLOCK_SIZE 512
+#define LFS_BLOCK_SIZE (64 * LFS_READ_SIZE)
 #endif

 #ifndef LFS_BLOCK_COUNT
-#define LFS_BLOCK_COUNT 1024
+#define LFS_BLOCK_COUNT 2096
 #endif

 #ifndef LFS_BLOCK_CYCLES
-#define LFS_BLOCK_CYCLES 1024
+#define LFS_BLOCK_CYCLES (1024 / 8)
 #endif

 #ifndef LFS_CACHE_SIZE
-#define LFS_CACHE_SIZE 64
+#define LFS_CACHE_SIZE 2048
 #endif

 #ifndef LFS_LOOKAHEAD_SIZE
-#define LFS_LOOKAHEAD_SIZE 16
+#define LFS_LOOKAHEAD_SIZE (4096 / 8)
 #endif
`

The same test failed for all branches:
`--- Split dir test ---
lfs_format: 0
lfs_mount: 0
: 0
Traceback (most recent call last):
  File "tests/test.py", line 61, in <module>
    main(*sys.argv[1:])
  File "tests/test.py", line 58, in main
    execute()
  File "tests/test.py", line 44, in execute
    subprocess.check_call(["./lfs"])
  File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['./lfs']' returned non-zero exit status -11`

As for our chip and read/program sizes.
We use two Micron MT29F4G08 on two different devices. Both share 2048 bytes page and blocks of 64 pages. One chip has 4096 blocks and another one 1024. Large files of 15 MB rely to the former.
512 byte is the smallest subpage size with ECC support. Sofar we try all page size and 1024 bytes and 512 bytes program size gives less errors in on-device tests with big files.