littlefs-project/littlefs #224: the problem of  file  size

Johnxjj, 2019-07-09:
boot_count = 1;
    lfs_file_rewind(&lfs, &file);
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));
    printf("size: %d\n", lfs_file_size(&lfs, &file));

    lfs_file_seek(&lfs, &file, 0, LFS_SEEK_END);
    boot_count = 2;
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));
    printf("size: %d\n", lfs_file_size(&lfs, &file));

    lfs_file_seek(&lfs, &file, 0, LFS_SEEK_END);
    boot_count = 3;
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));
    printf("size: %d\n", lfs_file_size(&lfs, &file));

    lfs_file_seek(&lfs, &file, 0, LFS_SEEK_SET);
    boot_count = 4;
    lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));
    printf("size: %d\n", lfs_file_size(&lfs, &file));
	
    // remember the storage is not updated until the file is closed successfully
    lfs_file_close(&lfs, &file);
    lfs_file_opencfg(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT, &file_cfg);
    for(int i=0;i<12;i+=4)
	{
		lfs_file_seek(&lfs, &file, i, LFS_SEEK_SET);
		lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));
		printf(" %d", boot_count);
	}
     printf("\n");
     lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));
     printf("size: %d\n", lfs_file_size(&lfs, &file));
``
the log :
boot_count: 0
size: 4
size: 8
size: 12
size: 12
4 4 4 
size: 4
boot_count: 4

Only the last size  is beyond my expectations. I think it should be 
"4 2 3
size: 12".
but the result is 
"4 4 4 
size: 4“.

lsilvaalmeida, 2019-07-09:
Hello!
I think this is the expected behavior. Like you haven't close your file before re-write from the begining, lfs will assumes that all the modifications made before are garbage. If you close your file before do the last seek and write, it should work like you want to. 
Your file have a size of 4, so when you try to read the position 4 and 8, lfs_file_read returns without any modifications to the buffer (here, boot_count). 

I hope that could be helpful.

geky, 2019-07-09:
Hi @Johnxjj, thanks for creating an issue. Also thanks @lsilvaalmeida, but he does call lfs_file_close on line 22.

@Johnxjj, I added a missing lfs_file_opencfg on line 1, but I'm not sure exactly how you opened the file. I ran this locally using emubd but it seems to be working for me, so your expectations are correct:

```
geky@s:littlefs$ ./lfs
lfs_format: 0
lfs_mount: 0
size: 4
size: 8
size: 12
size: 12
 4 2 3
size: 12
lfs_unmount: 0
```

What version of littlefs are you using? I wonder if the issue could have been fixed recently.
``` bash
git log --oneline -n 1
```

Also what is the configuration of `struct lfs_config` and `struct lfs_file_config`. These will be needed to reproduce this issue. It's possible attributes are somehow responsible.

Johnxjj, 2019-07-10:
@geky  i am sorry for that i miss  lfs_file_opencfg in here , but I have not forgotten it in my code. 
I use the V2.0 of littlefs. I downloaded it from git last month. Is it the latest version?

     
uint8_t lfs_read_buf[512];		                           
uint8_t lfs_prog_buf[512];		                            
uint8_t lfs_file_buf[512];		                         

__align(64) uint8_t lfs_lookahead_buf[32]; 

struct lfs_config cfg = 
{
	/* block device operations */														
	.read             = user_provided_block_device_read,   
	.prog             = user_provided_block_device_prog,   
	.erase            = user_provided_block_device_erase,  
	.sync             = user_provided_block_device_sync,   

	/* block device configuration */															
	.read_size        = 16,					         
	.prog_size        = 16,					         
	.block_size       = 4096,				              
	.block_count      = 256,				             
	.block_cycles     = 100,  		                       
	.cache_size       = 512,				              
	.lookahead_size   = 32, 		                   
	
	.read_buffer      = lfs_read_buf,		     
	.prog_buffer      = lfs_prog_buf,		        
	.lookahead_buffer = lfs_lookahead_buf,             
};

struct lfs_file_config file_cfg = 
{
	.buffer           = lfs_file_buf,				
};

Johnxjj, 2019-07-17:
@lsilvaalmeida Thank you for your reply！I tried it according to your method, and there will be problems. This problem occurs or not, depending on whether the file is stored in a super block or a data block.
@geky  After many tests, I came to a conclusion. 
       the inline file stored in the super block (only problems occur in the super block), if the file content is not modified at the end of the file (in the beginning and middle of the file) (and the size of the modified file does not exceed the end of the file), the modified file before the call close is normal, but after calling the close function, There will be an error in saving the tail pointer address of the file. Specifically, the tail pointer address points to the pointer position at the end of the modification (instead of the position of the previous file tail pointer), and the file length is also calculated according to this pointer. Therefore, after the close, the file data and the file length are abnormal.
      The read function illegally accesses content that exceeds the file size and returns the last data read in the file range. If the data has not been read before the file, the data at the end of the file is returned.
       In the above example, when I write a 4 to the 0 position of the file, everything is normal before the close, but after the close, the tail pointer points to the modified position "4" instead of "3". , so the file size becomes 4 instead of 12. In addition, in the process of reading later, the read function accesses the content exceeding the file size 4, and returns the data "4" read in the file range for the last time.

Johnxjj, 2019-07-17:
Hi @geky , sorry to bother you again. I want to know what version of littlefs are you using? Can you give me a version to solve the problem?

closed 2019-08-12 by Johnxjj