littlefs-project/littlefs #225: I do a software reset when the block was erased, which caused mount failed

Johnxjj, 2019-07-09:
`SPI_FLASH_WriteEnable();
			
  SPI_FLASH_WaitForWriteEnd();
  FLASH_SPI_CS_ENABLE();
  SPI_FLASH_SendByte(W25X_SectorErase);
  SPI_FLASH_SendByte((SectorAddr & 0xFF0000) >> 16);	
  SPI_FLASH_SendByte((SectorAddr & 0xFF00) >> 8);
  SPI_FLASH_SendByte(SectorAddr & 0xFF);	
  FLASH_SPI_CS_DISABLE();
	
  NVIC_SystemReset();

  SPI_FLASH_WaitForWriteEnd();` 

the  log :

lfs error:977: Corrupted dir pair at 0 1
lfs debug:3371: LINE: 3371   tag: -84
lfs debug:336: lfs_mount failed, mount again. ( err code: -84 )  
Corrupted
lfs debug:976: LINE: 976 
lfs error:977: Corrupted dir pair at 0 1
lfs debug:3371: LINE: 3371   tag: -84
lfs debug:356: lfs_mount failed again,over this code. ( err code: -84 )
Corrupted

I did an experiment: In order to simulate the power-down behavior, a software reset was added when the block was erased, which would cause the next mount to fail. I don't know why this happened.

geky, 2019-07-09:
Hi @Johnxjj, thanks for raising an issue. That shouldn't happen.

My first thought is to double check that W25X_SectorErase matches your configured block_size. If block_size is smaller or not a multiple of the physical erase size this could cause problems.

What configuration are you using (`struct lfs_config`)? Also what operations were you running during power-down? This info would help to reproduce the issue.

geky, 2019-07-09:
https://github.com/ARMmbed/littlefs/issues/222 may be related

Johnxjj, 2019-07-10:
@geky Thank you for your reply! I am sure that my configuration should be fine. And after my investigation, i am very happy that i solved it. The reason is that after the SPI is reset during the erase, reinitialize and continue working immediately, there may be some unexpected situation, which causes the data to fail to be read (as seen in the log, it is 0). I added a little delay in the initialization. At this time, it is possible to ensure the stable operation of the SPI.

log: 
lfs debug:521: sys_littlefs_test. LINE: 521 
lfs debug:105: SPIFLASH_Init. LINE: 105 
lfs debug:119: SPIFLASH_Init OK. LINE: 119 
lfs debug:329: lfs_mount. LINE: 329 
lfs debug:759: LINE: 759 
lfs debug:178: lfs read block:2000  off:0   size：16
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
lfs debug:178: lfs read block:2001  off:0   size：16
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
lfs debug:777: LINE: 777 
lfs debug:796: LINE: 796 
lfs debug:178: lfs read block:2000  off:0   size：512
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
lfs debug:805: LINE: 805  tag: 0 
lfs debug:818: LINE: 818  lfs_frombe32: 0

geky, 2019-07-18:
Ah! Good to hear. Does that mean this issue can be closed? I'll look into your other issues when I can.

closed 2019-07-19 by Johnxjj