littlefs-project/littlefs #228: Error "lfs debug:1469: Expanding superblock at rev 101" when writing to external SPI Flash

pjspanton, 2019-07-15:
I'm using an STM32F767 with external SPI Flash with external SPI Flash S25FL064L.
I'm logging every 1 second to Flash a single variable counter, all is OK for 25 to 26 minutes then I get the above error from littlefs.

I'm only able to start logging again by performing a full chip erase.    

Here are some details of my configuration:
Using latest master code.

 // block device operations
 cfg.read  = block_device_read;
 cfg.prog  = block_device_prog;
 cfg.erase = block_device_erase;
  cfg.sync  = block_device_sync;
// block device configuration,  total size is 8 Mbytes = 64 Mbits 
cfg.read_size = 256U;				//bytes
cfg.prog_size = 256U;				//bytes
cfg.block_size = 4096U;			//bytes
cfg.block_count = 2048U;			
cfg.block_cycles = 100U;	//Should be 100 to 1000, Num erase cycles before moving data to another block (wear levelling)
cfg.cache_size = 256U;				//bytes
cfg.lookahead_size = 256U;		//bytes

And every 1 second the following is called:
// read current Log counter (if file doesn't exist then create it)
    lfs_file_open(lfs, file, "LogCounter", LFS_O_RDWR | LFS_O_CREAT); 
    lfs_file_read(lfs, file, &log_count, sizeof(log_count));
    log_count += 1U;		// update count
    lfs_file_rewind(lfs, file);
    lfs_file_write(lfs, file, &log_count, sizeof(log_count));   
    lfs_file_close(lfs, file);  //storage is not updated until the file is closed successfully
    TRACE_PRINT("Log Counter: %d\n", log_count);    //TODO - debug only

It would appear that after the error (before I do a chip erase to rectify) we are getting stuck in lfs_dir_traverse() which I believe is a result of performing a close operation.  

I would much appreciate your help please.

pjspanton, 2019-07-15:
Further Test Results:
I have now tried disabling the Wear Levelling (by setting cfg.block_cycles = 0).
Logging has now been successfully running for 4 hours and still running fine (whereas previously I have never had it running for more than 26 minutes with cfg.block_cycles = 100).
I hope this additional evidence helps?

geky, 2019-07-15:
Hi @pjspanton, thanks for creating an issue. Do you know what the value of the counter was when it failed? I'm trying to reproduce locally but no luck yet, that geometry seems to work fine in simulation.

Also what hash of littlefs are you using?
`git log --oneline -n 1`

The "expanding superblock" message is a part of wear-leveling, so I'm not surprised disabling wear leveling removes the issue.

If littlefs notices the superblock has been erased too many times (cfg.block_cycles), and free blocks are available, it will grow a linked-list of blocks pointing to the actual superblock. That way we don't wear down the root of the filesystem early. I'm guessing the bug is caused by the interaction of this superblock-growth and other filesystem operations (the traverse?).

pjspanton, 2019-07-16:
Hi @geky ,  thanks for your reply.
I'm using the 4 source files from the latest Master commit as of July 2nd 2019 (abd90cb). 

Just ran the test again with wear leveling re-enabled (set to 100), details below at the point of error:
  Last Flash Write Addr (hex): 00001f00  
  Log Counter: 1580
  Timestamp: 2019-01-01 00-26-22  (ran for 26 minutes 22 seconds).

geky, 2019-07-16:
It's good that it's so consistent, but strange that I can't reproduce this. Two things may help:

1. I pushed a branch that has trace statements ([branch trace](https://github.com/ARMmbed/littlefs/tree/trace), will merge to master in future). Would you be able to capture the log with LFS_YES_TRACE defined? That will hopefully let me recreate what's exactly what's happening.

2. If it's possible, could you extract the image of the storage when it gets stuck? That may be enough to figure out what happened.

pjspanton, 2019-07-17:
Hi Geky, thanks for the suggestions, no. 1 is more straight forward so Iv'e attached the trace log here
as requested which uses your trace branch.  

[littlefs_trace_log_001.txt](https://github.com/ARMmbed/littlefs/files/3401635/littlefs_trace_log_001.txt)

pjspanton, 2019-07-17:
I've just performed some further tests, Please NOTE for this test Wear Leveling = 0 (disabled).
In addition to updating the Log Counter every 1 second, I am now (also every 1 sec) adding to a further file "DataLogFile_0001" where each time I write (append) 2 structures thus:
	lfs_file_open(lfs, file, "DataLogFile_0001", LFS_O_RDWR | LFS_O_CREAT | LFS_O_APPEND); 	
	lfs_file_write(lfs, file, &logRecordHeader, sizeof(logRecordHeader));
	lfs_file_write(lfs, file, &logSolarData, sizeof(logSolarData));
	lfs_file_close(lfs, file);

So the DataLogFile_0001 file grows in size each time, unfortunately this only lasts 15 seconds before it stops.

Firstly am I doing something wrong here with my sequence:  open, write, close.
Secondly, I feel I have an underlying issue here as wear leveling is disabled in this test.

Please find attached the log file for this test.

[littlefs_trace_log_002.txt](https://github.com/ARMmbed/littlefs/files/3402453/littlefs_trace_log_002.txt)

pjspanton, 2019-07-18:
Just been running some more tests on my SPI flash, writing to all sectors and reading back, all seems OK (this is before I mount Littlefs).

One further question:  in lfs.h I've obviously provided read/prog/erase support but what should I provide for "sync" as I'm not sure exactly what this should perform, see below:    
   // Sync the state of the underlying block device. Negative error codes
    // are propogated to the user.
    int (*sync)(const struct lfs_config *c);

geky, 2019-07-18:
Thanks for the logs, unfortunately it's not very clear what's going on and will take more time to investigate.

If you're able to extract the state of the block device when it's stuck, that would help quite a bit. Something like this would let you extract it over serial:
``` c
for (int i = 0; i < cfg.block_count; i += 1) {
    for (int j = 0; j < cfg.block_size; j += cfg.read_size) {
        uint8_t buffer[cfg.read_size];
        int err = cfg.read(&cfg, i, j, buffer, cfg.read_size);
        LFS_ASSERT(err == 0);
        for (int k = 0; k < cfg.read_size; k += 8) {
            printf("%x:%x: %x %x %x %x  %x %x %x %x\n", i, j + k,
                    buffer[k+0], buffer[k+1], buffer[k+2], buffer[k+3],
                    buffer[k+4], buffer[k+5], buffer[k+6], buffer[k+7]);
        }
    }
}
```

> One further question: in lfs.h I've obviously provided read/prog/erase support but what should I provide for "sync" as I'm not sure exactly what this should perform, see below:

I do need to provide better porting documentation (https://github.com/ARMmbed/littlefs/issues/56).

Sync is useful if you are caching in your block device, otherwise it can just be a noop:
``` c
int bd_sync(const struct lfs_config *c) { return 0; }
```

It's called at any synchronization points, when the filesystem requires that the physical disk reflects any caches to prevent corruption in the case of power-loss.

Imagine a program that writes application checksums to littlefs. It calls file open->write->close, and then after closing the file assumes that the checksum is safely stored on disk and it can overwrite the application in internal flash. If the block device driver has device-side caches for better performance, this assumption may not longer be true. If power is lost you could end up in a state where the application was updated but the checksum was never flushed to disk.

Hopefully that helps clarify what the sync function is for.

pjspanton, 2019-07-22:
Hi @geky , Thank you for the details on Sync, my sync function already just returns 0, so this would appear to be OK in my case.

As Requested, please find attached log file of the state of the block device when it gets stuck. 
Just to confirm, in this capture test I'm updating the LogCounter file and appending to the DataLogFile_0001 file every 1 second. The LogCounter only reaches 13 when it falls over.
Wear levelling = 100.   
I do hope this will highlight where it's going wrong.   

[lfs_stateofblockdevice.zip](https://github.com/ARMmbed/littlefs/files/3417454/lfs_stateofblockdevice.zip)

pjspanton, 2019-07-25:
Hi, any thoughts on my issue please, having sent over the captured "state of block device" log previously?

pjspanton, 2019-08-01:
Hi @geky, 
Thought I would migrate to v2.1 littlefs, still with the same problems.

pjspanton, 2019-08-06:
Can close this issue, traced to problems with one of my OS stacks, my apologies.

closed 2019-08-06 by pjspanton

geky, 2019-08-06:
Ah! Glad to hear you figured it out. Sorry I wasn't able to look into it earlier.