littlefs-project/littlefs #241: Porting LittleFS to internal Data flash on MCU

umanayana, 2019-07-24:
**Background**

- I have an NXP MCU that uses a flash module called FlexNVM. FlexNVM can be configure for Data Flash, this Data Flash is internal to MCU. 

- I currently have my own file system that uses a simplistic approach to wear leavering and data storage but as the project is becoming more demanding I think LittleFS will offer greater benefit for future expansion. 

**Question**

- I want to know how I could go about porting LittleFS to work with this Data Flash?

- What is the best way to come up with the optimal configuration for the parameters in `lfs_config`structure?

**Data Flash Specification**

- Size 256KBytes
- Sector size 4KBytes (I can only earse a sector at a time)

-

LaurentLouf, 2019-08-02:
The only thing you have to do is to port a block device implementation compatible with your hardware, there is no work needed on Little FS side. You can check for example https://github.com/ARMmbed/spif-driver that implements block device for SPI flash. If you remove the SPI related functions, you'll notice that it basically comes down to implement three functions : `read` , `program` and `erase`

techNiv1, 2020-08-24:
Hello @umanayana ,
have u found answer of your question?
I am also going through the same thing. Please share me some reference if u got  it solved.

Thanks in adv.

breakersun, 2021-08-28:
@techNiv1 just found a project based on STM32F1XX. The git history is in Chinese however I think it's pretty straightforward to understand the code hierarchy. Hope it helps you to clear your thoughts.

https://github.com/hanjhu328/LittleFSForSTM32F103