littlefs-project/littlefs #248: Infinite cycle problem

as564335sa, 2019-07-30:
Hi,
My hardware is STM32 + W25Q64
File system mounting and other functions work well, But when i repeatedly write a series of data in the same location of the same file. After 12 cycles, the program can not exit after entering `lfs_file_close`, and after reset at this time, `lfs_mount` will also loop indefinitely，and have to erase the flash sector to get back to normal. I write the following test code:
when `i = 0x0c` , lfs_file_close will not exit，I checked the driver and found no problem；
```
for(i = 0;i < 50;i++)
	{
		err = lfs_file_open(&lfs, &file, "/cfg", LFS_O_RDWR | LFS_O_CREAT);
		if((err != LFS_ERR_OK) && (err != LFS_ERR_EXIST))
		{
			//device_info.selected_name[0] = 0;
			break;
		}
		lfs_file_rewind(&lfs,&file);
		err = lfs_file_write(&lfs,&file,"hello lfs",9);
		lfs_file_close(&lfs,&file);
	}
```

as564335sa, 2019-07-30:
This the port:
```
int flash_read(const struct lfs_config *c, lfs_block_t block,
            lfs_off_t off, void *buffer, lfs_size_t size)
{
	SPI_Flash_Read(0,buffer,c->block_size * block + off,size);
	return 0;
}
int flash_write(const struct lfs_config *c, lfs_block_t block,
            lfs_off_t off, const void *buffer, lfs_size_t size)
{
	SPI_Flash_Write_NoCheck(0,(uint8_t*)buffer,c->block_size * block + off,size);
	return 0;
}
int flash_erase(const struct lfs_config *c, lfs_block_t block)
{
	SPI_Flash_Erase_Sector(0,block);
	return 0;
}
int flash_sync(const struct lfs_config *c)
{
	return 0;
}

const struct lfs_config cfg = {
    // block device operations
    .read  = flash_read,
    .prog  = flash_write,
    .erase = flash_erase,
    .sync  = flash_sync,

    // block device configuration
    .read_size = 256,
    .prog_size = 256,
    .block_size = 4096,
    .block_count = 2048,
    .cache_size = 128,
    .lookahead_size = 256,
};
```

Johnxjj, 2019-07-30:
Size of block caches must be a multiple of the read and program sizes, and a factor of the block size.

closed 2019-08-02 by as564335sa