littlefs-project/littlefs #252: misbehavior of `lfs_file_truncate`

pabigot, 2019-07-30:
It appears that lfs_file_truncate() moves the file head without recording that fact.

To reproduce: In a file system with 16 byte block size, and 64 byte cache, create a file that has 64 bytes increasing from 0x00 to 0x3F.  Close the file.

Then open the file and execute these commands:
```
uint8_t buffer[64];
printf("tell %u\n", (unsigned int)lfs_file_tell(lp, fp));
printf("read %d\n", lfs_file_read(lp, fp, buffer, sizeof(buffer)));
printf("\t%02x %02x %02x %02x\n", buffer[0], buffer[16], buffer[32], buffer[48]);
printf("seek %d\n", (int)lfs_file_seek(lp, fp, 0, LFS_SEEK_SET));
printf("tell %u\n", (unsigned int)lfs_file_tell(lp, fp));
printf("truncate %u\n", (unsigned int)lfs_file_truncate(lp, fp, 48));
printf("tell %u\n", (unsigned int)lfs_file_tell(lp, fp));
memset(buffer, -1, sizeof(buffer));
printf("read %d\n", lfs_file_read(lp, fp, buffer, sizeof(buffer)));
printf("\t%02x %02x %02x %02x\n", buffer[0], buffer[15], buffer[16], buffer[47]);
printf("tell %u\n", (unsigned int)lfs_file_tell(lp, fp));
```

The annotated output I get is:
```
tell 0      # file is correctly positioned at start after open
read 64     # file reads 64 bytes that appear to be numbered correctly
        00 10 20 30
seek 0      # Move back to start of file works
tell 0      
truncate 0  # Truncate to 48 bytes succeeds
tell 0      # File position is said to be zero
read 48     # Read returns the length of the file...
        30 3f 70 ff  # .. but the data is the 16 bytes after truncation,
                     # plus garbage past the original end of file
tell 48     #
```

Two anomalous behaviors:
* I did not expect lfs_file_truncate() to move the file position.  If that is expected behavior, `lfs_file_tell()` should have reported it.
* I did not expect to receive data that started at the truncated part of the file, and included garbage beyond the original file end.

Johnxjj, 2019-07-31:
@pabigot  You can set the file size to be larger than 513 bytes and try again. I think this should be no problem.

pabigot, 2019-07-31:
@Johnxjj I'm sure there are combinations of file lengths, truncation position, and cache size where it works.  The point is it does not work for these sizes.

Johnxjj, 2019-07-31:
@pabigot  Files smaller than 512 bytes are saved as inline files and stored in the super block. After my test, there are a lot of problems with inline files, so I want you to try other blocks.

pabigot, 2019-07-31:
If you start with an 8 KiBy file, truncate to 48, and read 64 bytes, you get exactly the same behavior: the returned data starts with data that had been at offset 48 and is correct for only 16 bytes (the read size), instead of starting with data that should be at 0 and being correct for 48 bytes.

If you start with an 8 KiBy file, truncate to (1024 + 48) (leaving the first 1 KiBy in place, a read of 64 bytes produces data that is correct for the first 64 bytes of the file.

This suggests the bug occurs whenever you truncate to a nonzero but small file length.  Presumably the data is moved into the cache, but the metadata is inconsistent.

Hope that helps.  It's still a pretty serious problem.

pabigot, 2019-08-03:
> Files smaller than 512 bytes are saved as inline files and stored in the super block.

It appears it's actually files no larger than `cfg->cache_size` that are saved as inline files.

AFAICT `cache_size` is stored in the superblock.  I'm not sure what happens if somebody mounts a file system with inline files using a config with a smaller `cache_size`.  @geky is that a problem?

geky, 2019-08-07:
Hey @pabigot thanks for putting up a fix for this.

> It appears it's actually files no larger than `cfg->cache_size` that are saved as inline files.

Yep, this is how littlefs coalesces files into a single block. And yes, `prog_size`, `read_size`, and `cache_size` are not needed for portability and are not saved in the superblock.

Getting this to work with inline files was an absolute nightmare. I was originally storing an `inline_size` in the superblock for this purpose but was able to figure out how to work around it.

The reason this was so important is because given that these are embedded devices, available RAM is going to be the biggest issue when considering portability between devices. And if I move a filesystem between a PC and an MCU, I want to be able to take advantage of however much RAM the system can through at me.

> AFAICT `cache_size` is *not* stored in the superblock. I'm not sure what happens if somebody mounts a file system with inline files using a config with a smaller `cache_size`. @geky is that a problem?

The only reason inline files are limited by the `cache_size` is because of the semantics of writing to multiple files at the same time.

It's a bit of a weird consideration for an embedded filesystem, but it was easy to solve in v1 when all files were COW linked-lists, so here we are now.

inlines files > `cache_size` is handled lazily as littlefs comes across them.

Writing to inline files that are larger than cache_size is easy, we just need to evict the file and move it to a skip-list on the first write. This is a part of the same logic that handles un-inlining files during normal operation:
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L2847

Reading inline files is a bit more complicated. We _really_ don't want to modify the filesystem during a read, since this rules out a readonly version and it could be confusing for users to get an out-of-space error during a read. 

But we also don't need to fit the entire file in the cache in this case. So what we do is load the file in chunks as we read. Each time we need a new chunk we fetch it from the directory. This is handled by `lfs_dir_getread` and `lfs_dir_getslice`:
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L2775

It's reasonable to think inline files > `cache_size` will cause a problem during compaction as we copy inline files to a new block, but `lfs_dir_compact` already handles a number of tags that are larger than `cache_size` by streaming them byte-by-byte and letting the cache logic chunk up the read/writes:
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L1214

Buuut, there was one last nasty tricky bit of behavior that can cause problems.

What if we have a file open for reading, and open the same file for writing, modify it, and then save. What happens to the open-for-reading file handle?

With files in COW skip-lists, this was easy. we update the pointer in the directory, but because it only contains on-disk-metadata, we still have a pointer to the skip-list in our open-for-reading file handle. The block allocator checks open files for in-use blocks, so the user can continue to read from the file.

But with inline files when we update the on-disk file, we may lose the old data. Normally this old data would be saved in the open-for-reading file handle's cache, but if our inline file is too large we can't do that.

I was stuck here for a while and was resigned to just have filesystems with large inline-files error until I realized something. At the moment of updating the metadata, someone, somewhere, is writing!

We can abuse this to evict the inline file. Sure there is the risk this might cause an OOS error at a weird time, but with wear-leveling there's already a lot of operations under the hood with even simple operations.

This is handled by this check in `lfs_dir_commit`:
https://github.com/ARMmbed/littlefs/blob/db054684a67956c5b86101dd70018accf284be2f/lfs.c#L1666-L1682

Anyways sorry for the long ramble. There was quite a bit of thought put into making sure inline files can remain portable.

Though there is the unfortunate fact that there's currently no _tests_ covering portability. This needs work. So take the above with a grain of salt.

Johnxjj, 2019-08-08:
@pabigot  My problem  #224  has been solved in the V2.1 version before, but in the test, I also encountered the same problem as you.

PR #257 (2019-08-08): fix seek position corruption in truncate function
[ftruncate](https://pubs.opengroup.org/onlinepubs/009695399/functions/ftruncate.html) is not supposed to move the seek pointer, but when making a file smaller the littlefs code did this.

The third patch in this set closes #252.

The first two are cleanup patches that (IMO) make it easier for people to understand the code and work with the code: one changing magic numbers to meaningful symbols, and one ensuring the file `off` field is not left uninitialized during open.  I can remove either or both of those if they're not wanted.
closed 2019-08-08 by geky