littlefs-project/littlefs #255: [Question] Can you determine the starting block for the FS?

joel-felcana, 2019-08-01:
Hi,

I understand that you can limit how much your FS extends into your flash by setting the block_count parameter, but how can you determine the starting block? My device writes the OTA update binaries to raw flash (as they must be parsed by the bootloader), and uses flash addresses 0 to 512KB. If I just erase them before an OTA, I (potentially) lose the FS. It's not a dealbreaker, I can start with a fresh FS after an update because I assume that an OTA will cause data loss, but I was just wondering, because I swear I read something about multiple, concurrent FS, but I can't find it now.

e107steved, 2019-08-01:
https://github.com/ARMmbed/littlefs/issues/176
https://github.com/ARMmbed/littlefs/issues/204

joel-felcana, 2019-08-02:
Thanks! I knew I read about it somewhere...

closed 2019-08-02 by joel-felcana