littlefs-project/littlefs #258: File operations should return LFS_ERR_BADF rather than asserting

ironss, 2019-08-04:
In commit 4850e01e14373d9808b78f3fb7833e26583744c1, code to check that the file was opened in the correct mode (read or write) was changed to asserts. The justification was that this the client code is wrong, and also that the code can be removed by disabling asserts.

Please will you re-consider this decision, and also consider changing the check that the file is actually open, so that they return LFS_ERR_BADF rather than asserting.

* All other filesystems I know of return an error code rather than asserting.
* There is no way for the client code to recover from this error.
* It creates a lot of extra book-keeping for the client to avoid the assert.
* The behaviour of my program changes when asserts are disabled.
* LFS must protect itself against client error, even with asserts disabled. What will LFS do if the client tries to write to a file opened for reading when asserts are disabled? If the client tries to close a file twice?

I have made the changes in my copy, and can create a pull request.

geky, 2019-08-05:
Hi @ironss, thanks for creating an issue, the feedback is always appreciated.

I'm open to reverting 4850e01, but I'm not convinced yet that returning LFS_ERR_BADF is valuable.

Unlike, say, LFS_ERR_NOSPC, LFS_ERR_BADF only occurs if the code is wrong. There's nothing in the environment outside of the code itself that can cause a file to be used while open in the wrong mode.

It's not a situation the user needs to be able to recover from, because the user is solely responsible for causing this situation to happen. There shouldn't be additional book-keeping other than "don't do that".

The behavior shouldn't change when you disable asserts, the assert prevents incorrect code from being written (as much as possible).

> LFS must protect itself against client error, even with asserts disabled. What will LFS do if the client tries to write to a file opened for reading when asserts are disabled? If the client tries to close a file twice?

This isn't entirely true. It's true we should protect against all disk errors. But incorrect file mode is entirely on the device side. And there's a limit to what we can protect against.

If the user modifies our struct directly, or casts pointers to different values, we can't protect against that. It's undefined behavior that the client just shouldn't do.

Consider this code, we simply _can't_ protect against this:
``` c
lfs_file_t file; // uninitialized, may be 1s, may be 0s
lfs_file_close(&lfs, &file); // undefined behavior
```

I'm all for adding as many asserts as possible to help prevent mistakes. But at some point it's a balancing act of if the code cost is worth additional logic to protect against incorrect usage. And since this targeting MCUs and small embedded systems, I lean towards code savings when possible.

---

At least, that's my perspective. Sorry it ended up kinda rambly. I'd still be happy to change incorrect mode usage to return LFS_ERR_BADF if I'm in the minority.

The question I have is what's the use case? From what I understand the only situations where you would need to be able to recover from incorrect mode usage is if you're either writing a CLI or porting to a programming language.

Johnxjj, 2019-08-14:
At the moment I have made some wrong operations, which will cause the program to die in LFS. I don't know if this is because I disabled the assertion. Sometimes I don't know that some operations are wrong. I hope that these error actions can be written in the function description, which will reduce the error.