littlefs-project/littlefs #265: File corruption and block-count assertion error on v2

felipeLeast, 2019-08-09:
I'm testing littlefs implementation in MBed OS, and I started seen a recurrent problem regarding block count assertion. The problem doesn't seem to follow a specific sequence:

`Error Message: Assertion failed: head >= 2 && head <= lfs2->cfg->block_count`

Another problem that's even more recurrent is that my program "looses" some files along the code. For example my code log follow:

```
[INFO][meas]: File /fs/measurement_80 has 623 bytes
lfs2 info:407: file_close(0x20039e80)
lfs2 info:409: file_close -> 0
[INFO][meas]: ========================================
lfs2 info:391: file_open(0x0, "measurement_config", 0x2)
lfs2 info:393: file_open -> 0
lfs2 info:474: file_size(0x20039e80)
lfs2 info:476: file_size -> 70
lfs2 info:452: file_seek(0x20039e80, 0, 0)
lfs2 info:454: file_seek -> 0
lfs2 info:419: file_read(0x20039e80, 0x20039fd0, 1024)
lfs2 info:421: file_read -> 70
[DBG ][conf]: Current index: 80, Types: 13
lfs2 info:407: file_close(0x20039e80)
lfs2 info:409: file_close -> 0
[DBG ][meas]: Measurement control: (Configurations: 13) 
lfs2 info:391: file_open(0x0, "measurement_80", 0x2)
lfs2 info:393: file_open -> -2
lfs2 info:391: file_open(0x0, "measurement_80", 0x602)
lfs2 info:393: file_open -> 0
lfs2 info:474: file_size(0x20039e80)
lfs2 info:476: file_size -> 0
lfs2 info:452: file_seek(0x20039e80, 0, 0)
lfs2 info:454: file_seek -> 0
lfs2 info:452: file_seek(0x20039e80, 0, 2)
lfs2 info:454: file_seek -> 0
[INFO][meas]: File /fs/measurement_80 currently has 0 bytes
```

What is happening in this code, I've appended on the file /fs/measurement_80 some bytes, then on the next cycle when I went to read it again it says that currently has 0 bytes. This happens very often.

Some times I'm reading an old file as well. I have a configurations file that I increment de index number, and when I'm on index 80 sometimes it read 30 (an example of old file) and on the next cycle it reads correct again.

My setup:

Hash:
I've started using LFS v1 which is the default from mbed os, than I changed to V2. From: https://github.com/ARMmbed/mbed-littlefs

SPI FLash Block Device: 
MBED_LFS2_BLOCK_SIZE: 4096
MBED_LFS2_BLOCK_CYCLES: 1024
MBED_LFS2_CACHE_SIZE: 64
MBED_LFS2_LOOKAHEAD_SIZE: 64

Flash device:
W25Q16JVSNIQ

MCU:
NRF52840

felipeLeast, 2019-08-13:
Following up on this error. I was doing some other test and during that, the code listed all my directories, I then I saw the bad behaviour. It's not only corrupting my files, but changing it's name. Follows a print from https://github.com/ARMmbed/mbed-os-example-filesystem output:

```
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    boot_coun
    con
    measurement_confi
    location
    schedule_1_5153430
    schedule_1
    measurement_0
    measurement_1
    measurement_2
    measurement_3
    measurement_4
    measurement_5
    measurement
    measurement_7
    measurement_8
    measurement_9
    measuremen
    measure
    measurement_11
    measurement_12
    measur
    measurement_14
    measurement_15
    measurement_16
    measurement_17
    measurement_18
    measurement_19
    measurement_2
    measurement_21
    measurement_22
    measurement_23
    m
    measurement_25
    measurement_26
    config
    measurement_27
    measuremen
    measurement_config
    measurement_6
    measurement_10
    measurement_13
    boot_count
    measurement_29
    measurement_30
    measurement_31
    measurement_32
    measurement_33
    measurement_34
    measurement_35
    measurement_36
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    11
    12
    13
    14
    15
    16
    17
    18
    19
    20
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```

Notice that there's a lot of files with incomplete naming, for example, all measurement files should follow an index such as "measurement_0", so "measure" is incorrect, and no where in my code I change or use this name. 

Does anyone have any idea what may be causing this?

Johnxjj, 2019-08-14:
I see you file_open(0x0, "measurement_80", 0x602), and then file_size(0x20039e80). You said you have appended on the file /fs/measurement_80 some bytes, but I don't see the any write and close. Could you give more details ？

felipeLeast, 2019-08-14:
Sorry @Johnxjj I think that my log snnipet wasn't very complete. But I saw that the problem it's probably not in littleFS but in the integration with mbed-os's Wi-SUN, maybe something to do with multiple SPI's being used. I tried swtiching to FAT FS and still code the same error. I'll keep looking into it.