littlefs-project/littlefs #268: No error but file corrupt with lot of 0xFF in between data

ajaybhargav, 2019-08-12:
Hi @geky,

lfs has been acting strange. There is no error during read/write but file data itself is corrupt with lot of 0xFF in between correct data. I dumped the file (using lfs API, not raw SPI) on console, so you can see reading file is working fine. But the data inside is corrupted. See below:

The actual data... and then bunch of 0xFF
[image]

after 3220 bytes I get the correct data again:
[image]

The correct data starts with header, AIS/AHTT and ends with length of data written which is 0x70 in this case as highlighted above. We have only 4 files in the filesystem to store datalog. The current error case, sizes are as follows. both normal_pri.bin and normal_sec.bin are corrupt.
0byte critical_pri.bin
232bytes critical_sec.bin
36804bytes normal_pri.bin
37180bytes normal_sec.bin
What I also observed is file also contains information or root record (metablock?) how can that be possible? it should not be part of file at all. see the file atually starts with this:

[image]

Any help would be appreciated.

Regards,
Ajay Bhargav

lsilvaalmeida, 2019-08-12:
Hello @ajaybhargav. I think understand your "problem". In fact, I think that is the expected behaviour. In littlefs, a file takes at least one block (if not an inline file). So your file really end with 0x70 (see the last word befor the Fs, in little endian). After the end of your file, the rest of the block is "empty" that mens, normally, that all bits are set to 1. 

The others files are in others blocks with the same structure. 

I hope this helps.

ajaybhargav, 2019-08-12:
Hi @lsilvaalmeida I think you missed the point. I mentioned clearly the dump images i posted is actual data of **file** not **SPI block data**. So I do not expect RAW SPI block data to show up in files. I know the data i write ends with 0x70 (I mentioned that) but I do not expect to see 0xFF in it. What **file data** should show is continuous stream of
[header][data][len][header][data][len]
like this... but something else is happening.
[image]

I did another test of writing data to two log files at 5 sec and 10sec interval respectively. The corruption ended up getting me two files with same file size and with same data. Which is not expected. as both files have different frequency of writing so cannot have same size.

ajaybhargav, 2019-08-13:
I must explain what I do here, I store logs in file system. The logs are appended to a log file with following structure:
{ [header] [CRC] [Data] } [Len of data in {}]
The data is appended one after the other. Now when I retrieve the logs, I delete them from file on successful retrieval. Based on number of logs retrieved I truncate the file to length of data read. I am able to read 100 to 150 logs.. sometimes even more. But retrieval stops after some time (not fixed count). The moment it stops, I end up with a corrupted file. I have no idea why file truncate causes file to corrupt. Maybe it has to do something with the way file is organized during truncation.

Johnxjj, 2019-08-14:
Can you give a simple code example to locate the problem quickly.

ajaybhargav, 2019-08-14:
Here is a sample code.
[lfs.zip](https://github.com/ARMmbed/littlefs/files/3501478/lfs.zip)

Zip contain 2 file:
lfs_write.c -> generates packet
Command line to test:
`./lfs_write <packets> [format]`
packets: Num of packets, default 50
format: optional, format lfs before use.

lfs_read.c -> read, verify and delete packet using truncate.
Command line:
`./lfs_read`

Problem starts anywhere from 30 packets. try it once or twice and you will get corrupt len error.

ajaybhargav, 2019-08-16:
Did you get a chance to test this?

ajaybhargav, 2019-08-22:
There some issue in truncation logic when file goes beyond block size. As a workaround I changed my logging code, So instead of writing in a single file. I write into multiple files while keeping the file size under 4K (block size). I do not see any issue while doing truncate on file. Please fix this issue.

ryankurte, 2020-05-07:
We were observing what _appears_ to be the same issue, where truncate is used on long files sometimes the dirent or the whole file system end up corrupted.

After a bunch of fumbling around, increasing the cache and lookahead sizes both from 16 bytes to 128 bytes seems to have alleviated this in our case (although we have not _exhaustively_ tested this yet).

Indumathi1412, 2020-05-11:
I am facing file size resets, (file truncates) after file block size reached. How it can be solved?
[image]

ajaybhargav, 2020-05-11:
> 
> 
> Here is a sample code.
> [lfs.zip](https://github.com/ARMmbed/littlefs/files/3501543/lfs.zip)
> 
> Zip contain 2 file:
> lfs_write.c -> generates packet
> Command line to test:
> `./lfs_write <packets> [format]`
> packets: Num of packets, default 50
> format: optional, format lfs before use.
> 
> lfs_read.c -> read, verify and delete packet using truncate.
> Command line:
> `./lfs_read`
> 
> Problem starts anywhere from 100 packets. try it once or twice and you will get corrupt len error.
> 
> **Edit:**
> Wrong files attached earlier. Updated now.

@ryankurte I have provided sample code to test. It might be old.

@Indumathi1412 I am keeping filesize under 4K (block size for SPI flash I used) and that workaround is working well for me as of now. Try to do the same and see if it works.

Indumathi1412, 2020-05-12:
I kept 1K file size, beyond the size made file truncates but my physical memory address keeps on increasing, after it reaches max address, file corruption occurs as no more space is available.  My requirement is not to immediately read the log data. Recent 10K(10 files each 1K size) log data is enough. Is it possible to write, the file from the start address. I tried lfs_remove() same behaviour. I am using Nor Flash IC.
[image]

hgspbs, 2020-06-26:
truncate extends from wrong block

[t.patch.txt](https://github.com/ARMmbed/littlefs/files/4836595/t.patch.txt)

geky, 2023-04-17:
Hi all, thanks for creating an issue and the detective work. Sorry about the delay in addressing this.

This was definitely a fundamental issue in `lfs_file_truncate` and @hgspds's solution is correct.

I've created a PR for this here: https://github.com/littlefs-project/littlefs/pull/800, let me know if anyone sees anything wrong with it.