littlefs-project/littlefs #270: Rename corrupts files

winston-todd, 2019-08-14:
I've found that if I rename a file while another file is open, any new data that I try to write to the open file is instead written to the renamed file.

Here is a test case that triggers the bug:

    echo "--- Rename test ---"
    scripts/test.py << TEST

        lfs_size_t size;

        lfs_format(&lfs, &cfg) => 0;
        lfs_mount(&lfs, &cfg) => 0;

        // open a log file, write something, then close it
        lfs_file_open(&lfs, &file, "log_0", LFS_O_CREAT | LFS_O_WRONLY) => 0;
        lfs_file_write(&lfs, &file, "0000000000", 10) => 10;
        lfs_file_close(&lfs, &file) => 0;

        // open test file
        lfs_file_open(&lfs, &file, "test", LFS_O_CREAT | LFS_O_WRONLY) => 0;

        // write to test file
        lfs_file_write(&lfs, &file, "1111111111", 10) => 10;
        lfs_file_sync(&lfs, &file) => 0;

        // rename log file
        lfs_rename(&lfs, "log_0", "log_1") => 0;
        
        // write to test file
        lfs_file_write(&lfs, &file, "2222222222", 10) => 10;
        lfs_file_sync(&lfs, &file) => 0;

        // close test file
        lfs_file_close(&lfs, &file) => 0;

        // check log file contents
        lfs_file_open(&lfs, &file, "log_1", LFS_O_RDONLY) => 0;
        size = lfs_file_read(&lfs, &file, buffer, sizeof(buffer) - 1);
        buffer[size] = 0;
        printf("%s\n", buffer);
        test_assert(size, 10);
        strcmp((const char*)buffer, "0000000000") => 0;

        // check test file contents
        lfs_file_open(&lfs, &file, "test", LFS_O_RDONLY) => 0;
        size = lfs_file_read(&lfs, &file, buffer, sizeof(buffer) - 1);
        buffer[size] = 0;
        printf("%s\n", buffer);
        test_assert(size, 20);
        strcmp((const char*)buffer, "11111111112222222222") => 0;

    TEST

Johnxjj, 2019-08-20:
I'm sorry, i didn't see any error in you code.

winston-todd, 2019-08-20:
When I add the test case to test_files.sh and run make test_files, this is the output:

```
$ make test_files
./tests/test_files.sh
=== File tests ===
--- Rename test ---
11111111112222222222
./tests/test_files.sh:48: assert failed with 20, expected 10
    size => 10
```

The test case writes "0000000000" to "/log_0", then renames it to "/log_1", but reading the file after the rename gives "11111111112222222222".

I have run the test in both WSL x64 and a Fedora x64 virtual machine, and I get the same result.

Johnxjj, 2019-08-21:
I think even if you don't rename the file "log_0", the result is still "11111111112222222222". After you first write "1111111111" to the file "test", the pos of file will move to the end of file. Then you continue write "2222222222" to  the same file. The result will do become "11111111112222222222" . you need to use lfs_file_seek(... ,0,LFS_SEEK_SET).

winston-todd, 2019-08-21:
Right, but the problem is the contents of "log_1" is "11111111112222222222", when it should be "0000000000".

Johnxjj, 2019-08-21:
Yes，it is! I try it just now. But i think this woud be a illegal operation. User known how to avoid if. lfs_rename don't need open the file.

winston-todd, 2020-04-17:
I ran my test case with the v2.2 release and it passed, so I'm closing this issue.

closed 2020-04-17 by winston-todd