littlefs-project/littlefs #271: lfs: lfs.c:77: lfs_bd_read: Assertion `0' failed.

brezen, 2019-08-15:
index 969e198..c8d125c 100644
--- a/scripts/template.fmt
+++ b/scripts/template.fmt
@@ -51,11 +51,11 @@ char path[1024];
 #endif
 
 #ifndef LFS_BLOCK_SIZE
**-#define LFS_BLOCK_SIZE 512
+#define LFS_BLOCK_SIZE 4096**
 #endif
 
 #ifndef LFS_BLOCK_COUNT
**-#define LFS_BLOCK_COUNT 1024
+#define LFS_BLOCK_COUNT 256**
 #endif
 
 #ifndef LFS_BLOCK_CYCLES

**root@am:~/littlefs# make test_alloc**
./tests/test_alloc.sh
=== Allocator tests ===
--- Single-process allocation test ---
--- Multi-process allocation test ---
--- Single-process reuse test ---
--- Multi-process reuse test ---
--- Cleanup ---
--- Exhaustion test ---
lfs_error:475: No more free space 383
--- Exhaustion wraparound test ---
lfs_error:475: No more free space 302
--- Dir exhaustion test ---
lfs_error:475: No more free space 185
lfs_error:475: No more free space 184
--- Chained dir exhaustion test ---
lfs_error:475: No more free space 296
lfs_error:475: No more free space 295
lfs_error:475: No more free space 295
**lfs: lfs.c:77: lfs_bd_read: Assertion `0' failed.**
make: *** [test_alloc] Error 2

brezen, 2019-08-19:
Does nobody have this issue?

e107steved, 2019-08-19:
It's not at all obvious what you're trying to do when you get this assert.
However there is one very obvious thing - you've increased LFS_BLOCK_SIZE by a factor of 8, and decreased LFS_BLOCK_COUNT by a factor of 4. Could this be relevant?

brezen, 2019-08-20:
Yes, it has to do with changing the size of the block. Restoring the original will not have this problem.
But there's no choice.，erase size can‘t be 512
[image]

brezen, 2019-08-29:
any suggestion?

embeddedt, 2019-09-04:
Like @e107steved said, you haven't provided much information. But his point still stands:

> However there is one very obvious thing - you've increased LFS_BLOCK_SIZE by a factor of 8, and decreased LFS_BLOCK_COUNT by a factor of 4.

That means that (if I did the math correctly) you have cut the total size of the device in half. Therefore, the test is probably running out of space.

Obviously, though an assertion shouldn't fail when there is no space left.

mr-at-eo, 2019-09-19:
Same Problem, with 10M ramdisk using 512b Sectors:

    #define  LFS_READ_SIZE       LFS_BLOCK_SIZE
    #define  LFS_PROG_SIZE       LFS_READ_SIZE
    #define  LFS_BLOCK_SIZE      512
    #define  LFS_BLOCK_COUNT     (10*1024*1024)/LFS_BLOCK_SIZE
    #define  LFS_BLOCK_CYCLES    1024
    #define  LFS_CACHE_SIZE      LFS_BLOCK_SIZE
    #define  LFS_LOOKAHEAD_SIZE  128

    === Allocator tests ===                                                       
    --- Single-process allocation test ---                                        
    --- Multi-process allocation test ---                                         
    --- Single-process reuse test ---                                             
    --- Multi-process reuse test ---                                              
    --- Cleanup ---                                                               
    --- Exhaustion test ---                                                       
    lfs_error:472: No more free space 20644                                       
    --- Exhaustion wraparound test ---                                            
    lfs_error:472: No more free space 1426                                        
    --- Dir exhaustion test ---                                                   
    lfs_error:472: No more free space 20891                                       
    lfs_error:472: No more free space 20890                                       
    --- Chained dir exhaustion test ---                                           
    lfs_error:472: No more free space 20629                                       
    lfs_error:472: No more free space 20628                                       
    lfs_error:472: No more free space 20628                                       
    lfs: lfs.c:75: lfs_bd_read: Assertion `block < lfs->cfg->block_count' failed.