littlefs-project/littlefs #274: lfs_error:971: Corrupted dir pair at 0 1 AND lfs_warn:1637: Superblock 1 has become unwritable

XiaoBaiCX, 2019-08-20:
I got strange error at the  my software 
My Debug Message as flow：
[16:33:03.301]收←◆lfs_error:971: Corrupted dir pair at 0 1

[16:33:03.386]收←◆lfs_debug:1632: Bad block at 0
lfs_warn:1637: Superblock 1 has become unwritable

[16:33:03.426]收←◆lfs_error:971: Corrupted dir pair at 0 1

My code is as flow：
void Flash_APP(void)
{
    // mount the filesystem
    int err = lfs_mount(&lfs, &cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err) {
        lfs_format(&lfs, &cfg);
        lfs_mount(&lfs, &cfg);
   }
}
using latest Littlefs version

my config is :
const struct lfs_config cfg = {
    // block device operations
	
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync  = user_provided_block_device_sync,

    // block device configuration
    .read_size = 256,
    .prog_size = 256,
    .block_size = 4096,
    .block_count = 8192,
    .cache_size = 256,
    .lookahead_size = 1024,
};
If you know why. please help me!  thank you!

XiaoBaiCX, 2019-08-22:
reason is  i no init  lfs_config struct   { .block_cycles}member
then  The problem is solved ！  
thanks！

SaiPraveen22, 2022-06-09:
im getting the same error can you explain how you solved it, thank you